analogdevicesinc/gr-iio #1: problem compiling gr-iio

courcirc8, 2016-09-13:
Hello,
  I'm trying to compile gr-iio with pyboms (on Ubuntu). I compiled libiio and libad9361, but gr-io gives:
_____________________________________________________________
[  3%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o
/home/sdr-pc/gnu_radio/gr-iio/lib/device_source_impl.cc: In static member function ‘static iio_context* gr::iio::device_source_impl::get_context(const string&)’:
/home/sdr-pc/gnu_radio/gr-iio/lib/device_source_impl.cc:136:52: error: ‘iio_create_context_from_uri’ was not declared in this scope
       ctx = iio_create_context_from_uri(uri.c_str());
                                                    ^
lib/CMakeFiles/gnuradio-iio.dir/build.make:73 : la recette pour la cible « lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o » a échouée
make[2]: *** [lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o] Erreur 1
CMakeFiles/Makefile2:135 : la recette pour la cible « lib/CMakeFiles/gnuradio-iio.dir/all » a échouée
make[1]: *** [lib/CMakeFiles/gnuradio-iio.dir/all] Erreur 2
Makefile:127 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-iio:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gr-iio. Aborting.
_____________________________________________________________________
The manual compilation gives the same error.
Do you have an idea of what is going wrong?
Many thanks

pcercuei, 2016-09-13:
I think the version of libiio is too old - you should use the version 0.7 or later.

courcirc8, 2016-09-14:
Thanks for your answer. I'm using the last libiio version from analogdevice. Could you tell me where iio_create_usb_context is supposed to be defined? I can't find in it any of my libs with a grep -r
Thx.

courcirc8, 2016-09-17:
Paul, apparently you suppressed the support to create usb context in v0.8:
https://github.com/analogdevicesinc/libiio/commit/764f99b3781045789dd18d53b68b92aa2a7f5c92
With this revert, gr-iio will request libiio ver 0.7 because it was not updated.
I will try to intsall libiio ver 0.7...

pcercuei, 2016-09-20:
Libiio v0.8 does not exist, we're still at v0.7 right now.

Also we did not suppress the USB backend; we changed the way it's implemented (when it was still experimental).
The function iio_create_usb_context does not exist anymore. Now the function iio_create_context_from_uri is used (with an URI starting with "usb:").

courcirc8, 2016-09-21:
Apparently, the src files that get downloaded in my pybomb prefix directory are an older version of gr-iio, but updating the gnu-radio recipes did not solved that issue.
I'm just using
pybombs install gr-iio
here is what I find inside gr-recipes/gr-iio.lwr
Is there anything else I should look at?

category: common
depends:
- gnuradio
- libiio
- libad9361
description: GNU Radio IIO Blocks
gitbranch: gr-iio
inherit: cmake
satisfy:
  deb: gr-iio
source: git+https://github.com/analogdevicesinc/gnuradio.git

pcercuei, 2016-09-21:
Ok, looks like upstream GNU Radio devs screwed up, the recipe doesn't point to the right sources...

courcirc8, 2016-09-21:
I see... I tried to use the recipes from your repro /recipes but still without success (missing libiio and libad9361 recipes)

pcercuei, 2016-09-21:
I created a pull request with the changes needed to make it work:
https://github.com/gnuradio/gr-recipes/pull/60

courcirc8, 2016-09-21:
I modified the recipes locally, carefully removed libiio and now everything compiles nicely!
Many thanks!

closed 2016-09-21 by pcercuei