analogdevicesinc/gr-iio #4: Using multiple USB contexts causes 'ERROR: Unable to claim interface 5: -16'

tfcollins, 2017-08-17:
When setting blocks representing the same radio (source and sink) to both use USB contexts, one will claim the context and lock the other out.

One or both must use IP backend instead.

rgetz, 2017-08-17:
Don't we just have a single "radio" block? 2 blocks (even under IP) are problematic- setting different sample rates will cause issues.

tfcollins, 2017-08-17:
No. Everything is source or sink in our GR blocks.  Having a single block would cause the interfaces to wait for each other too.

mhennerich, 2017-08-18:
I run into the same problem only recently. Lars mentioned that there are provisions that both blocks can share the same context. But for some reason this is currently not working. Someone needs to have a look.

lclausen-adi, 2017-08-18:
It looks like those provisions are only in Scopy, not in the gr-iio blocks.

tfcollins, 2017-08-18:
Ok thanks Lars.  I'll take a look at the differences.

tfcollins, 2017-08-18:
@lclausen-adi is there a way to check used/connected contexts given uri and device name?

Basically we need to check if another block has an already open context and act appropriately.  Paul's manager in scopy sits on top of the flowgraph so it has knowledge of the blocks, but you can't really do this from within a block.

We may have to implement a base class where both source and sink blocks inherit from with some static members to keep track of these contexts.

rgetz, 2017-08-18:
@

rgetz, 2017-08-18:
My understanding is that since libiio is thread safe (so no shared memory), (and you could be trying to talk to the same iio context in multiple applications, from different pid/memory spaces) there isn't any way for libiio to do that robustly. 

It might work in the same thread/same application, but would fail in any other situation. 

Lars can correct me if I'm wrong. 

- Robin

lclausen-adi, 2017-08-21:
There is no global state within libiio itself, so libiio doesn't know about which contexts exist and which don't.

Now the question is can we somehow attach global state to the flowgraph from within a block so that it can be shared between multiple independent blocks.

tfcollins, 2017-08-21:
Blocks have no knowledge of the parent flowgraph, and therefore other blocks.  They are purposely designed this way.

A simple way I see to check for multiple block instances is through a static member in a parent class.

@Lars will blocks need to share a context with different devices or have two different contexts?

pcercuei, 2017-08-21:
Global state is just a static iio_context pointer protected by a mutex and usecount, no?

tfcollins, 2017-08-21:
Basically something like that shared between the device_source and device_sink blocks.

commit 30bda41f (2017-08-24): Added static map to keep track of requested contexts with same uris. fixes #4
PR #10 (2017-10-03): Claim interface bug
Fixes #4.  Tested with Pluto and FMComms2 board.  All context mode variations worked (usb,ip,blank).
closed 2017-10-03 by tfcollins