analogdevicesinc/gr-iio #5: Latest tag (pybombs recipe) does not work with PLUTO

mrjacobagilbert, 2017-08-25:
Not so much an issue as a recommendation: The latest gr-iio tag, v0.2, is specified in your [gr-iio PyBombs recipe](https://github.com/gnuradio/gr-recipes/blob/master/gr-iio.lwr) and does not include the ADALM-PLUTO specific blocks and anyone using PyBOMBS will not have access to the PLUTO blocks. I wanted to recommend you either retarget the PyBOMBS recipe at master or make a new tag to make it easier for GR/PyBOMBS usesrs to get up and running.

tfcollins, 2017-08-25:
There is an issue or two remaining before the next release.  Hopefully we can get this done soon thanks.

tfcollins, 2017-12-13:
PR has been opened to fix this: https://github.com/gnuradio/gr-recipes/pull/103

tfcollins, 2018-01-11:
PR has been merged

closed 2018-01-11 by tfcollins