analogdevicesinc/gr-iio #6: Add versioning requirements for Bison

tfcollins, 2017-08-31:
On some distro's like centos bison is version 2.7 in their mirrors, which causes build failures.  3.0.4 works but not sure if earlier releases are valid.

rgetz, 2017-08-31:
You are right -- Centos7.3-1611 seems to point to 2.7 (which is ancient - released Dec 2012)
https://centos.pkgs.org/7/centos-x86_64/bison-2.7-4.el7.x86_64.rpm.html

Bison 3.0.4 was released Jan 2015. Time for them to upgrade.
http://www.rpmfind.net/linux/rpm2html/search.php?query=bison&submit=Search+...&system=&arch=

lclausen-adi, 2017-09-01:
I believe anything >= 3.0 is good. We might be able to re-write the grammar to be compatible with older versions, but I don't think we want to do that.

And it is only needed for the iio-math module (which isn't iio related at all), we could add an option to disable it, or even move it to its own gr-math package.

tfcollins, 2017-09-02:
Yes you are correct Lars on the version.  I do agree that just adding the version requirement is a better solution.

Maybe in the future we can look at partitioning those math blocks off (all 3 of them).

commit 8dcae1df (2017-09-29): Merge pull request #8 from analogdevicesinc/add-bison-versioning

Added requirement for bison version.  This fixes #6.
commit e4739544 (2017-09-29): Added requirement for bison version.  This fixes #6.
closed 2017-09-29 by tfcollins