analogdevicesinc/gr-iio #7: Enabling Pluto's FPGA decimation filter

tfcollins, 2017-09-14:
Currently (I believe) we are not using the additional dec8 filter in Pluto's FPGA.  This needs to be added for certain rates from the mask.  Maybe we can get this from the filter designer as well when it gets merged into libad9361-iio: https://github.com/analogdevicesinc/libad9361-iio/tree/filter_generation

tfcollins, 2018-09-24:
This is being added to libad9361, so little change will be required here.

mobilinkd, 2020-12-23:
What is the status of this enhancement request?

tfcollins, 2020-12-23:
This enhancement is part of our upstream PR to GR itself: https://github.com/analogdevicesinc/gnuradio/blob/merge-griio-3.9/gr-iio/lib/device_source_impl.cc#L454

If you need something now I would enable the filters manually through the sampling_frequency attribute in the cf-ad9361-lpc and cf-ad9361-dds-core-lpc drivers.

-Travis