analogdevicesinc/gr-iio #11: ImportError: cannot import name iio

drom, 2017-10-04:
OpenSuSe Tumbleweed.
Followed install instruction on https://wiki.analog.com/resources/tools-software/linux-software/gnuradio

Instantiated Pluto Source --> Qt GUI Frequency Sink
F5 --> Ok
F6 -->
```
Executing: /usr/bin/python2 -u /home/drom/work/github/drom/pluto/top_block.py

Traceback (most recent call last):
  File "/home/drom/work/github/drom/pluto/top_block.py", line 22, in <module>
    from gnuradio import iio
ImportError: cannot import name iio
```

tfcollins, 2017-10-04:
What happens when you run "import gnuradio.iio" from a python terminal?

pcercuei, 2017-10-05:
I think you need to compile and install gr-iio with -DCMAKE_INSTALL_PREFIX=/usr, otherwise the Python scripts will not be found by the Python runtime, as it does not search in /usr/local by default.

drom, 2017-10-07:
@pcercuei Thank you very much. It works.
Does it mean the change in instruction? or it is just my case?

pcercuei, 2017-10-07:
I think the instructions could be changed, yes.

tfcollins, 2017-10-09:
Updated the wiki for this case: https://wiki.analog.com/resources/tools-software/linux-software/gnuradio?&#compiling_the_gr-iio_blocks_for_your_pc

closed 2017-10-09 by tfcollins

mhennerich, 2017-10-10:
Please approve wiki pages after updating them. If for some reason you can't approve them please let me know. I approved this one for you.