analogdevicesinc/gr-iio #15: Decimation and interpolation parameter removal

tfcollins, 2017-11-29:
I think we may want to remove these parameters from the block masks from FMComms and Pluto since they can cause aliasing based on how they are implemented.

https://github.com/analogdevicesinc/gr-iio/blob/2997f46a18823c4d20f6d2ca9481c2caa75394f8/lib/device_sink_impl.cc#L162
https://github.com/analogdevicesinc/gr-iio/blob/2997f46a18823c4d20f6d2ca9481c2caa75394f8/lib/device_source_impl.cc#L237

Is there a strong use case for them?

mhennerich, 2017-11-30:
It was introduced with this commit:
https://github.com/analogdevicesinc/gr-iio/commit/8b9eaccd196717f35e073498069ebf9f500a60fd

I don't remember the reason - but simply skipping samples in SW is not going to make a strong use case. This can be better done outside the block.

At some point we were talking about adding support for the Pluto Int/Dec by 8 HW filter.
But this is something different.

lclausen-adi, 2017-11-30:
It was done to safe CPU resources so some of the designs were capable of running on the ARM processor on the ZYNQ.

tfcollins, 2017-11-30:
I'm fine with leaving it for the iio_device blocks, but having it in the FMComms/Pluto blocks is just asking for trouble.  I think this was more of a concern before the baseband_auto_rate function existed, but now we can get down to much lower rates.  Especially if we add the hooks to control Pluto's FPGA filters

tfcollins, 2017-12-12:
Are the FMComms/Pluto blocks required for scopy?  Doing a quick search of the repo I couldn't find anything.

lclausen-adi, 2017-12-12:
no

commit 162ddd85 (2017-12-12): Merge pull request #16 from analogdevicesinc/remove-intdec

Removed instances of interpolation and decimation in fmcomms2/3/4/5 a…
Fixes #15
closed 2017-12-12 by tfcollins