analogdevicesinc/gr-iio #20: Shared context get destroyed too early

tfcollins, 2018-02-16:
If a context is shared between blocks it should destroy only by the last running block using it.

rgetz, 2018-02-16:
and this is primarily due to libiio not keeping track of it's own users, and forcing everyone to do their own context manager (since we don't want to add the requirement for atomic operations (ie boost) in libiio)?

lclausen-adi, 2018-02-16:
We have platform independent mutual exclusion support in libiio (https://github.com/analogdevicesinc/libiio/blob/master/lock.c). That's not the problem.

rgetz, 2018-02-16:
so, why do we want to avoid counting references to a context then? (sorry if we went over this before).

tfcollins, 2018-02-17:
This was part of a bug we fixed awhile ago (https://github.com/analogdevicesinc/gr-iio/issues/4). However, when the contexts were being destroyed it needs to be done in a specific order.  Basically the last block with it open needs to do the destroying.

PR #21 (2018-02-28): Context closure bug
This fixes #20 as well as fixes an appveyor build issue.  Now we keep track of all open contexts, the number of blocks using each context, and their uris.

This came about from an issue: https://ez.analog.com/thread/103178-runtimeerror-unable-to-create-buffer-16-after-added-fmcomms2-blocks-on-no-gui-run-to-completion-gnuradio
closed 2018-02-28 by tfcollins