analogdevicesinc/gr-iio #23: ImportError: No module named iio_swig

drom, 2018-03-20:
Anybody can help me with this issue?
OpenSuSe Tumbleweed
```
Traceback (most recent call last):
  File "/home/drom/work/github/drom/OFDM/kiss_tx_ofdm.py", line 27, in <module>
    from gnuradio import iio
  File "/usr/lib64/python2.7/site-packages/gnuradio/iio/__init__.py", line 35, in <module>
    from iio_swig import *
ImportError: No module named iio_swig
```

tfcollins, 2018-03-20:
How did you install gnuradio on your system and the gr-iio blocks?

drom, 2018-03-20:
I used OpenSuSe package manager zypper to install gnuradio
And I compiled gr-iio block like described here: https://wiki.analog.com/resources/tools-software/linux-software/gnuradio?&#compiling_the_gr-iio_blocks_for_your_pc

tfcollins, 2018-03-20:
You probably have a path issue then.  Is /usr in your PYTHONPATH?

drom, 2018-03-20:
Sorry. I found an issue.
I was missing `python-devel` package.
`sudo zypper in python-devel` -- fixed it.
@tfcollins  thank you for the help.

closed 2018-03-20 by drom

ThisChessPlayer, 2018-07-02:
I am currently running into a similar issue using a Raspberry Pi and PlutoSDR setup. I compiled the gr-iio blocks the same way as drom and got same error message when using PlutoSDR Source/Sink blocks in gnuradio. I think my paths are set up correctly, but I am pretty new to some aspects of this so cannot be sure. 
PYTHONPATH=/usr 
LD_LIBRARY_PATH=/usr/local/lib

tfcollins, 2018-07-02:
@ThisChessPlayer are the generated SWIG python blocks included in your PYTHONPATH?

ThisChessPlayer, 2018-07-02:
I have also tried with combinations of PYTHONPATH=/usr/lib/python2.7/dist-packages/, /usr/lib/python2.7/site-packages, and /usr/lib, but it does not make a difference.

If I go into iio_swig.py and remove the try/except around the import, I get a more detailed import error: /usr/lib/libgnuradio-iio.so.0: undefined symbol: _ZN2gr6blocks14transcendental4makeERKSsS3_

This is the current file structure I have, which appears to match what I have on my Ubuntu 16.04 setup which is working:

/usr/lib/python2.7/dist-packages/gnuradio/iio
iio_pluto_sink_swig.py/pyc/pyo
_iio_pluto_sink_swig.so
iio_pluto_source_swig.py/py/pyo
_iio_pluto_source_swig.so
iio_swig.py/pyc/pyo
_iio.swig.so
__init__.py/pyc/pyo

/usr/lib/python2.7/site-packages/
iio.py
iio.pyc
libiio-0.15-py2.7.egg-info

/usr/lib/
libgnuradio-iio.so
libgnuradio-iio.so.0
libgnuradio-iio.so.0.3

tfcollins, 2018-07-02:
Can you provide your install_manifest.txt file from the build directory?

ThisChessPlayer, 2018-07-02:
I decided to go with Ubuntu Mate image for the Raspberry Pi, and everything worked smooth as butter.
For reference though, the install_manifest.txt file contained the following:
/usr/lib/pkgconfig/gnuradio-iio.pc
/usr/lib/cmake/iio/iioConfig.cmake
/usr/include/gnuradio/iio/api.h
/usr/include/gnuradio/iio/device_source.h
/usr/include/gnuradio/iio/device_sink.h
/usr/include/gnuradio/iio/fmcomms2_source.h
/usr/include/gnuradio/iio/fmcomms2_sink.h
/usr/include/gnuradio/iio/fmcomms5_source.h
/usr/include/gnuradio/iio/fmcomms5_sink.h
/usr/include/gnuradio/iio/pluto_source.h
/usr/include/gnuradio/iio/pluto_sink.h
/usr/include/gnuradio/iio/converter_ss.h
/usr/include/gnuradio/iio/math.h
/usr/include/gnuradio/iio/power_ff.h
/usr/include/gnuradio/iio/modulo_ff.h
/usr/include/gnuradio/iio/modulo_const_ff.h
/usr/lib/libgnuradio-iio.so.0.3
/usr/lib/libgnuradio-iio.so.0
/usr/lib/libgnuradio-iio.so
/usr/lib/python2.7/dist-packages/gnuradio/iio/_iio_swig.so
/usr/lib/python2.7/dist-packages/gnuradio/iio/iio_swig.py
/usr/lib/python2.7/dist-packages/gnuradio/iio/iio_swig.pyc
/usr/lib/python2.7/dist-packages/gnuradio/iio/iio_swig.pyo
/usr/lib/python2.7/dist-packages/gnuradio/iio/_iio_pluto_source_swig.so
/usr/lib/python2.7/dist-packages/gnuradio/iio/iio_pluto_source_swig.py
/usr/lib/python2.7/dist-packages/gnuradio/iio/iio_pluto_source_swig.pyc
/usr/lib/python2.7/dist-packages/gnuradio/iio/iio_pluto_source_swig.pyo
/usr/lib/python2.7/dist-packages/gnuradio/iio/_iio_pluto_sink_swig.so
/usr/lib/python2.7/dist-packages/gnuradio/iio/iio_pluto_sink_swig.py
/usr/lib/python2.7/dist-packages/gnuradio/iio/iio_pluto_sink_swig.pyc
/usr/lib/python2.7/dist-packages/gnuradio/iio/iio_pluto_sink_swig.pyo
/usr/include/swig/iio_swig.i
/usr/include/swig/iio_pluto_source_swig.i
/usr/include/swig/iio_pluto_sink_swig.i
/usr/lib/python2.7/dist-packages/gnuradio/iio/__init__.py
/usr/lib/python2.7/dist-packages/gnuradio/iio/__init__.pyc
/usr/lib/python2.7/dist-packages/gnuradio/iio/__init__.pyo
/usr/share/gnuradio/grc/blocks/iio_device_source.xml
/usr/share/gnuradio/grc/blocks/iio_device_sink.xml
/usr/share/gnuradio/grc/blocks/iio_fmcomms2_source.xml
/usr/share/gnuradio/grc/blocks/iio_fmcomms2_sink.xml
/usr/share/gnuradio/grc/blocks/iio_fmcomms5_source.xml
/usr/share/gnuradio/grc/blocks/iio_fmcomms5_sink.xml
/usr/share/gnuradio/grc/blocks/iio_pluto_source.xml
/usr/share/gnuradio/grc/blocks/iio_pluto_sink.xml
/usr/share/gnuradio/grc/blocks/iio_math.xml
/usr/share/gnuradio/grc/blocks/iio_math_gen.xml
/usr/share/gnuradio/grc/blocks/iio_power_ff.xml
/usr/share/gnuradio/grc/blocks/iio_modulo_ff.xml
/usr/share/gnuradio/grc/blocks/iio_modulo_const_ff.xml

tfcollins, 2018-07-02:
Sometimes you have to explicitly add /usr/lib/python2.7/dist-packages/gnuradio/iio to your PYTHONPATH.

ThisChessPlayer, 2018-07-03:
I ended up trying this to see what would happen, so PYTHONPATH=/usr/lib/python2.7/dist-packages/:/usr/lib/python2.7/dist-packages/gnuradio/iio, made sure with echo $PYTHONPATH, ran gnuradio-companion, and ran into same error. It could have just been me running it off of Raspbian, which I suspect is not officially supported. I should have been more specific about that! Anyways, thanks for your support. I have it running well with Ubuntu Mate :)

tfcollins, 2018-07-04:
I spun up my Pi.  Its seem like the repo installs of GR for the Pi get put in a weird place.  Other OOT's I tried failed to load as well in the default path.  I had to manually copy the python files over into the nonlocal folder like: 

`cp -r /usr/local/lib/python2.7/dist-packages/gnuradio/iio /usr/lib/python2.7/dist-packages/gnuradio/`

That worked for me.

nepoch046, 2019-01-30:
I decide this problem by coping new image of rootfs, without gr-iio, and try install. Because if you installed any time gr-iio with problem, you won't install success

trxstudio, 2020-05-11:
Adding

add_definitions(-DBOOST_CHRONO_HEADER_ONLY)

to the CMakeLists.txt file in upgrade-3.8 branch of gr-iio allows me
to import iio from python without error.

https://www.mail-archive.com/discuss-gnuradio@gnu.org/msg70829.html