analogdevicesinc/gr-iio #24: Removing gnuradio from python installation path

f--t, 2018-03-31:
Currently the CMakeFile for python installs in 

```(CMake)
GR_PYTHON_INSTALL(
    FILES
    __init__.py
    DESTINATION ${GR_PYTHON_DIR}/gnuradio/iio
)
```
which makes it impossible manage the PYTHON_PATH for gnuradio independently of gr-iio. It seems

```
GR_PYTHON_INSTALL(
    FILES
    __init__.py
    DESTINATION ${GR_PYTHON_DIR}/iio # <-- no gnuradio
)
```
might be a better choice as it complies with other gr overlays.

f--t, 2018-03-31:
I can make a pull request with the changes if that's of any interest.

tfcollins, 2018-04-02:
I've created a branch to do this (will push once I get off the airplane), but there are two main aspects that need to be considered first.  gr-iio gets used in our scopy tool and this may break those pieces, which we don't want to do.  We are also considering trying to merge gr-iio upstream if possible, which is why the code is structured the way it is.  So I can't totally guarantee we'll merge an update.

@f--t is the current installation structure breaking something for you?  I just want to better understand the use case here.  Are you using gr-iio outside gnuradio?

tfcollins, 2018-04-03:
This branch should meet your needs for now: https://github.com/analogdevicesinc/gr-iio/tree/python-install-change

f--t, 2018-04-04:
Thanks for the update, the back-story for this request comes from trying to build gr-iio for Nixos. The way Nix manages python dependencies, and hence GNU-Radio is via manipulation of the PYTHONPATH env variable; e.g. ```/nix/store/f7xqbacrj7cj5yfd7babcrnv93r0xpni-gr-iio/lib/python2.7/site-packages```. Since currently both gr and gr-iio use the same directory structure for their libraries, there is a path collision and the python interpreter fails to find the right modules for gr-iio and import fails.

f--t, 2018-04-04:
The changes proposed in the branch "python-install-change" fixes those issues.

tfcollins, 2018-04-12:
This change would conflict with our libiio python bindings since they share the same module name (iio), so we cannot make this change.

closed 2018-04-12 by tfcollins