analogdevicesinc/gr-iio #28: iio context not being destroyed after program termination

djcopley, 2018-05-17:
I am looking at the output waveforms on my o-scope and after the program quits, and the LO frequency remains after the program terminates. I am 100% positive it is not frequency just noise from the oscillator since it is outputting at a strong -13dBm. My thought is for some reason the fmcomms2_sink module is not properly terminating? Possibly the context is never being destroyed? These tones go away after reinitializing the kernel (echo 1 > /sys/kernel/debug/iio/*insert remainder of file path*/initialize). Furthermore, there are sometimes 3 distinct peaks visible after the program has terminated, each about 100MHz apart from each other (usually slightly quieter then the center frequency).

rgetz, 2018-05-18:
The radio is sort of separate from the IIO context. 

When the IIO context exits, the hardware stays in the last state the software set it. It doesn't automatically stop Tx (or Rx).

djcopley, 2018-05-18:
Is it possible to have it destroyed when the program exists?

rgetz, 2018-05-18:
First - IIO scope doesn't use the gr blocks, so, no - not there. 

The 2nd part is it's up to the application to do that, it's a user space/application function, not a hardware or library function. From the hardware standpoint - we don't know when something is "done". When a context is destroyed - there isn't anything that notifies the remote (that I'm aware of), since things like network context can be open simultaneously by multiple applications. It would be bad if something "restored" or "stopped" things while it was being used by something else.

djcopley, 2018-05-18:
@mhennerich @rgetz Thanks for the help. That makes a lot more sense.

closed 2018-05-18 by djcopley