analogdevicesinc/gr-iio #31: Windows Usage?

evandavey, 2018-06-27:
Can this be used with windows gnuradio (installed from the binary package)?  Ideally as a binary but could build as well if that's an option.

tfcollins, 2018-06-27:
Yes gr-iio will build fine on windows, as well as all the necessary dependencies.

evandavey, 2018-06-27:
Ok thanks @tfcollins.  Is there a build guide or instructions I can follow?  What are the dependencies?

tfcollins, 2018-06-28:
The generic build guide is here: https://wiki.analog.com/resources/tools-software/linux-software/gnuradio

I would suggest starting here: https://github.com/gnieboer/GNURadio_Windows_Build_Scripts
and inserting the gr-iio build process

Here is also how we build gr-iio on appveyor which may be helpful for binary dependency links: https://github.com/analogdevicesinc/gr-iio/blob/master/appveyor.yml

evandavey, 2018-06-28:
Hmm ok thanks.  Looks like it's going to be pretty involved and not sure I have the time for that right now.  Would your appveyor binary work?  It would be great if a binary version could be supplied so that it easy to use the Pluto with GNURadio purely in windows as the WSL approach has issues (e.g. audio sink, USB support e.g. for RTL-SDR).

tfcollins, 2018-06-28:
Getting GNURadio to work at all on Windows is a challenge.  The community itself doesn't really support it, so its kinda hard for us to.

The appveyor build isn't really complete and is just meant for some smoke tests.  I don't believe the GUI functions at all.

evandavey, 2018-06-28:
OK thanks.  It's a shame as it does detract from the user-friendliness of the Pluto product, particularly given the windows binary seems to work well (even if not officially supported).  WSL is close (and better than a Linux desktop VM IMO) but not quite ready for prime-time.  PyBOMBs looks to be unnecessarily convoluted versus using standard package management for binaries and/or compiling from source for those wanting bleeding edge.

closed 2018-06-28 by evandavey