analogdevicesinc/gr-iio #37: Fail to run Pluto with Gnuradio

alexanderthefox, 2018-09-22:
Hello!

I installed gnuradio and the iio-blocks which are described at https://wiki.analog.com/resources/tools-software/linux-software/gnuradio on my ubuntu 18.04.01.
After loading the fmradio_pluto.grc I tried to run it. 
But I received the following message:
Loading: "/home/alexander/gr-iio/iio-examples/fmradio_pluto.grc"
>>> Done

Generating: '/home/alexander/gr-iio/iio-examples/top_block.py'

Generating: '/home/alexander/gr-iio/iio-examples/top_block.py'

Executing: /usr/bin/python -u /home/alexander/gr-iio/iio-examples/top_block.py

Traceback (most recent call last):
  File "/home/alexander/gr-iio/iio-examples/top_block.py", line 15, in <module>
    from gnuradio import iio
ImportError: cannot import name iio

Pls, can someone explain to me, what went wrong an how I could overcome that problem.

Looking forward to hearing from you.

Thx
Alexander

alexanderthefox, 2018-09-22:
Yes, I build and installed libiio.
Is there a way to check if the installation of libiio was ok?

tfcollins, 2018-09-22:
There can be a namespace issue sometimes with apt installs of gnuradio.  Please read the last section here: https://wiki.analog.com/resources/tools-software/linux-software/gnuradio#compiling_the_gr-iio_blocks_for_your_pc

alexanderthefox, 2018-09-22:
Tfcollins, I' m not quite sure if I understand you? 
I am using Ubuntu 18.04.1 not Windows.
I followed the mentioned installation guide to you linked.

tfcollins, 2018-09-22:
Sorry, missed that you copied things to /usr/lib as suggested.  Do you have swig installed?

mhennerich, 2018-09-23:
You need install and build things in the proper order. So now go back an rebuild libiio

alexanderthefox, 2018-09-23:
Mhennerich,

should I just reinstall libiio?
Is it neccessary to delet the current installed libiio or can I just compile libiio ?

tfcollins, 2018-09-23:
If you didn't have swig when you built gr-iio, then it didn't build the python blocks. Rebuilding gr-iio should be enough.

alexanderthefox, 2018-09-23:
Tfcollins,

ok. Just compiling again without removing before building, right?

alexanderthefox, 2018-09-24:
@all:
Many thx for your help. 
Gnuradio is now running with pluto!

closed 2018-09-24 by tfcollins