analogdevicesinc/gr-iio #42: ImportError: No module named _iio_swig

christophjurczyk, 2019-04-27:
Hi,
I am trying to include an IIO device in Gnuradio, but have some issues with the iio package. I installed the package on my Manjaro as follows:

> git clone https://github.com/analogdevicesinc/gr-iio.git
> cd gr-iio
> cmake -DCMAKE_INSTALL_PREFIX=/usr .
> make 
> sudo make install
> cd ..
> sudo ldconfig 

When I try to include iio in python, following error occurs:

> -> from gnuradio import iio
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/usr/lib/python2.7/site-packages/gnuradio/iio/__init__.py", line 35, in <module>
>     from iio_swig import *
>   File "/usr/lib/python2.7/site-packages/gnuradio/iio/iio_swig.py", line 17, in <module>
>     _iio_swig = swig_import_helper()
>   File "/usr/lib/python2.7/site-packages/gnuradio/iio/iio_swig.py", line 16, in swig_import_helper
>     return importlib.import_module('_iio_swig')
>   File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
>     __import__(name)
> ImportError: No module named _iio_swig

When I add the package to the `PYTHONPATH`, I get following error:

> -> import sys
> -> print sys.path
> ['', '/usr/lib/python27.zip', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-linux2', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/usr/lib/python2.7/site-packages', '/usr/lib/python2.7/site-packages/gtk-2.0', '/usr/lib/python2.7/site-packages/wx-3.0-gtk3']
> -> sys.path.append("/usr/lib/python2.7/site-packages/gnuradio/iio")
> -> from gnuradio import iio
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/usr/lib/python2.7/site-packages/gnuradio/iio/__init__.py", line 35, in <module>
>     from iio_swig import *
>   File "/usr/lib/python2.7/site-packages/gnuradio/iio/iio_swig.py", line 17, in <module>
>     _iio_swig = swig_import_helper()
>   File "/usr/lib/python2.7/site-packages/gnuradio/iio/iio_swig.py", line 16, in swig_import_helper
>     return importlib.import_module('_iio_swig')
>   File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
>     __import__(name)
> ImportError: libad9361.so.0: cannot open shared object file: No such file or directory

Does anyone have an idea how to fix this issue? 
Many thanks in advance.
Christoph

tfcollins, 2019-04-29:
Make sure the libad9361 libraries are on path as well, or just move them to /usr/lib/.
gr-iio should be available in the AUR as well.

christophjurczyk, 2019-05-06:
Thanks for your help. I copied the compiled libad9361 library to /usr/lib/libad9361.
When I now try in import it to python, my complete python environment crashes. Any ideas?

`Python 2.7.16 (default, Mar 11 2019, 18:59:25) 
[GCC 8.2.1 20181127] on linux2
Type "help", "copyright", "credits" or "license" for more information.
-> from gnuradio import iio
Segmentation fault (core dumped)`

In gnuradio I get still the same error. I installed the gnuradio block with the help of [package](https://aur.archlinux.org/packages/gnuradio-iio-git/) with an arch linux package manager. But this returns the same error in gnuradio.

tfcollins, 2019-05-30:
My guess is that its loading things incorrectly and/or cannot find headers. The only way to really debug it would be to enable debug symbols and run gdb.

mariobaldini, 2019-05-31:
Hello, 

I am facing a similar issue in a Fedora 30 environment. 
```
(...)
  File "/usr/lib64/python2.7/site-packages/gnuradio/iio/iio_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_iio_swig')
  File "/usr/lib64/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _iio_swig

```

*Installation steps:*
```
cd ~/code
git clone https://github.com/analogdevicesinc/libiio.git
cd libiio
cmake -DLIB_SUFFIX=64 .
make 
sudo make install
cd ..

cd ~/code
git clone https://github.com/analogdevicesinc/libad9361-iio.git
cd libad9361-iio
cmake -DLIB_SUFFIX=64 .
make 
sudo make install
cd ..

cd ~/code
git clone https://github.com/analogdevicesinc/gr-iio.git
cd gr-iio
# cmake -DCMAKE_INSTALL_PREFIX=/usr .
cmake -DLIB_SUFFIX=64 -DCMAKE_INSTALL_PREFIX=/usr .

make 
sudo make install
cd ..
sudo ldconfig

cp -r /usr/local/lib64/python2.7/site-packages/gnuradio/iio /usr/lib64/python2.7/site-packages/gnuradio

``` 

I tried the lib copy indicated in the wiki (`cp -r /usr/local/lib/python2.7/dist-packages/gnuradio/iio /usr/lib/python2.7/dist-packages/gnuradio/`) just updating the lib64 and site-packages fields, as they seemed to be different in this evironment. 

Although I still can not fix and not sure what is wrong. 

Could someone please advise what should I do?

```
echo $PATH
/home/mario/.local/bin:/home/mario/bin:/usr/share/Modules/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin
```

`/usr` contents: https://www.dropbox.com/s/wnbqbui47e5hcdf/list-usr.txt?dl=0
Installed packages: https://www.dropbox.com/s/gd79h6wnrv6t8ek/list-installed-packages.txt?dl=0

Thank you
Mario

tfcollins, 2019-06-01:
For Fedora you will likely have to add the python blocks to your PYTHONPATH. Try explicitly add the gnuradio/iio folder

mariobaldini, 2019-06-01:
@tfcollins Indeed, I needed to add it to the PYTHONPATH:
`echo "export PYTHONPATH=/usr/lib64/python2.7/site-packages/gnuradio/iio" >> ~/.bashrc`

and also copy libad9361 to the proper path, to fix another error after that:
`sudo cp /usr/local/lib/libad9361.so* /usr/local/lib64`
(although it was built with the proper sufix `cmake -DLIB_SUFFIX=64 .`, this copy was also necessary).

christophjurczyk, 2019-06-07:
I solved my ImportError. First of all I switched to the current version of Ubuntu and followed the [installation guide](https://wiki.analog.com/resources/tools-software/linux-software/gnuradio).
In addition I had to execute following commands:

> cp /usr/include/gnuradio/swig/*.i swig/
> cp /usr/include/gnuradio/swig/gnuradio_swig_bug_workaround.h swig/

Thanks for your advises.

closed 2019-06-07 by christophjurczyk

mariobaldini, 2019-06-08:
PS: On Fedora I also needed to copy the libad9361 to the following location:
`sudo cp /usr/local/lib64/libad9361* /usr/lib64/`

Not sure that is the exactly single place for it in Fedora x64, but it works now.