analogdevicesinc/gr-iio #47: Gain Mode options of FMComms2 source in GNU Radio

alvionnes, 2019-07-04:
Hello!

I would like to know the difference between the Gain Mode options (Manual, Slow, Fast, Hybrid) of FMComms2 source in GNU Radio. Do they have different parameter values? If yes, what parameters are different and what are the values?

If you can lead me to a link that would answer my question, that would be great. Thanks in advance!

tfcollins, 2019-07-05:
See here: https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361?s[]=fastlock#gain_control_modes
Additional information is available in the UG-570 document: https://www.analog.com/en/design-center/landing-pages/001/integrated-rf-agile-transceiver-design-resources.html

alvionnes, 2019-07-09:
Thanks tfcollins! I'm a newbie so thanks for helping. :)

closed 2019-07-09 by alvionnes