analogdevicesinc/gr-iio #55: upgrade-3.8: module 'iio' has no attribute 'pluto_sink'

Cyancali, 2019-10-07:
Hi,
I ran PyBOMBS to install Gnuradio 3.8 and the analogdevices modules. Therefore I compiled libiio, libad9361 from master branch and gr-iio from upgrade-3.8 branch. Everythink ran fine and also all the blocks are in GRC available. (Ubuntu 18.04, PyBOMBS Version 2.3.3, Prefix Python version is: 2.7.15)

But as soon as I run python and call iio.pluto_sink(...) block (or other blocks in the iio repo) this error message pops up:
AttributeError: module 'iio' has no attribute 'pluto_sink'

I don't really know what is happening. Running either Python2 or Python3 does not change anything. In both Python versions I can import iio, but calling pluto_sink fails.

Does anyone has a clue what is going on?

tfcollins, 2019-10-08:
Make sure the generated swig module is on path. When you run `sudo make install` it prints all the files it installs and where for gr-iio

Cyancali, 2019-10-08:
Not sure whether I understood you correctly. The module gets installed in my pybombs prefix. Snippet of `sudo make install`:
```
-- Installing: /home/user/pybombs_gnuradio/lib/python3.6/dist-packages/iio/iio_pluto_sink_swig.py
```

`PYTHONPATH` includes python3.6/dist-packages directory.

tfcollins, 2019-10-08:
`import iio` imports the libiio bindings not the gr-iio blocks. Remove iio.py from the python3.6/dist-packages folder

Cyancali, 2019-10-09:
Thank you for your answer! I was able to solve the issue with your help! But I didn't yet really understand how it happened as it was a fresh/clean install, but i could find an `iio.py` file in python2.7/site-packages. I deleted them and now it runs. For others who ran into the same problem, first check where the source file is located after import and if it matches with the output during calling 'sudo make install':

```
$ python3
>>> import iio
>>> iio.__file__
'/home/user/pybombs_gnuradio/lib/python2.7/site-packages/iio.py'
```
After checking what is in the directory, I deleted all *iio* files:
```
$ ll lib/python2.7/site-packages/
iio.py iio.pyc libiio-0.18-py2.7.egg-info
$ rm lib/python2.7/site-packages/*iio*
```

closed 2019-10-09 by Cyancali

eyobed-dev, 2023-12-22:
i solved it by just copying iio folder