analogdevicesinc/gr-iio #60: ModuleNotFoundError: No module named '_iio_swig'

paulgrahek, 2020-01-20:
Running on Pop! OS 19.10
Using GNU Radio 3.8.0.0 from the GNU Radio PPA.

Using the upgrade-3.8 branch, setup steps as follows worked correctly:
```
git clone https://github.com/analogdevicesinc/gr-iio.git
cd gr-iio
cmake .
make 
sudo make install
cd ..
sudo ldconfig
```
However, when running a flow graph with a Pluto source block, the following error(s) occurs:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/iio/iio_swig.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: dynamic module does not define module export function (PyInit__iio_swig)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/iio/__init__.py", line 30, in <module>
    from .iio_swig import *
  File "/usr/lib/python3/dist-packages/iio/iio_swig.py", line 17, in <module>
    _iio_swig = swig_import_helper()
  File "/usr/lib/python3/dist-packages/iio/iio_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_iio_swig')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_iio_swig'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/iio/iio_swig.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: dynamic module does not define module export function (PyInit__iio_swig)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/paulgrahek/Nextcloud/isu/EECPI/Lab 4/receiver_fm_2.py", line 38, in <module>
    import iio
  File "/usr/lib/python3/dist-packages/iio/__init__.py", line 36, in <module>
    from .iio_swig import *
  File "/usr/lib/python3/dist-packages/iio/iio_swig.py", line 17, in <module>
    _iio_swig = swig_import_helper()
  File "/usr/lib/python3/dist-packages/iio/iio_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_iio_swig')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_iio_swig'
```
Unlike the other module import issues, this one appears to have some sort of dynamic module error but I'm not sure what the root cause is.

tfcollins, 2020-01-21:
On my PopOS after the installation I simply added `/usr/local/lib/python3/dist-packages` to my PYTHONPATH and everything worked fine. What else did you do have running make install for the gr-iio blocks?

paulgrahek, 2020-01-22:
Figure it out. My PYTHONPATH was set to ```/usr/lib/python3/dist-packages:/usr/local/lib/python3/dist-packages```. Changing the order to ```/usr/local/lib/python3/dist-packages:/usr/lib/python3/dist-packages``` or setting it to only ```/usr/local/lib/python3/dist-packages``` resolves the issue.
Thank you @tfcollins!

closed 2020-01-22 by paulgrahek