analogdevicesinc/gr-iio #62: Issues making gr-iio

The-inside-man, 2020-01-26:
When running through the steps to make gr-iio I run into the following error..

Any thoughts on how to resolve?

jake@jake-Blade:~/gr-iio$ make
[  2%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o
In file included from /usr/local/include/gnuradio/block.h:27:0,
                 from /usr/local/include/gnuradio/sync_block.h:27,
                 from /home/jake/gr-iio/include/gnuradio/iio/device_source.h:27,
                 from /home/jake/gr-iio/lib/device_source_impl.h:30,
                 from /home/jake/gr-iio/lib/device_source_impl.cc:28:
/usr/local/include/gnuradio/basic_block.h:94:17: error: ‘rpcbasic_sptr’ was not declared in this scope
     std::vector<rpcbasic_sptr> d_rpc_vars; // container for all RPC variables
                 ^~~~~~~~~~~~~
/usr/local/include/gnuradio/basic_block.h:94:17: note: suggested alternative: ‘top_block_sptr’
     std::vector<rpcbasic_sptr> d_rpc_vars; // container for all RPC variables
                 ^~~~~~~~~~~~~
                 top_block_sptr
/usr/local/include/gnuradio/basic_block.h:94:30: error: template argument 1 is invalid
     std::vector<rpcbasic_sptr> d_rpc_vars; // container for all RPC variables
                              ^
/usr/local/include/gnuradio/basic_block.h:94:30: error: template argument 2 is invalid
lib/CMakeFiles/gnuradio-iio.dir/build.make:73: recipe for target 'lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o] Error 1
CMakeFiles/Makefile2:218: recipe for target 'lib/CMakeFiles/gnuradio-iio.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-iio.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

tfcollins, 2020-01-27:
What Linux distribution are you using? What gnuradio version are you building against and how was it installed?

The-inside-man, 2020-01-27:
Ubuntu 18.04.03 LTS, GNURadio 3.9.0.0-git (Python 3.6.9)

Installed via the command line. sudo apt-get install gnuradio

tfcollins, 2020-01-27:
So it looks like you are using the master branch PPA and not the standard Ubuntu package. What branch from gr-iio are you using?

The-inside-man, 2020-01-27:
you are correct... I am on master. What branch should I be on?

 master
  remotes/origin/HEAD -> origin/master
  remotes/origin/ad9371
  remotes/origin/appveyor
  remotes/origin/attr-block-update
  remotes/origin/buffer_size
  remotes/origin/cancel
  remotes/origin/clang-format
  remotes/origin/debian
  remotes/origin/debug_threaded_win
  remotes/origin/filter-designer
  remotes/origin/fix-cyclic-device-sink
  remotes/origin/grcon2019
  remotes/origin/grcon2019_m2k
  remotes/origin/master
  remotes/origin/matlab
  remotes/origin/no-phy-support
  remotes/origin/no_math
  remotes/origin/python-install-change
  remotes/origin/python-reorg
  remotes/origin/single-param
  remotes/origin/start_stop
  remotes/origin/static-libs
  remotes/origin/underflow-overflow-support
  remotes/origin/upgrade-3.8

tfcollins, 2020-01-27:
upgrade-3.8

The-inside-man, 2020-01-27:
seems gnuradio's cmake is behind?

-- Build type not specified: defaulting to release.
CMake Error at CMakeLists.txt:86 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.8".

  The following configuration files were considered but not accepted:

    /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.9.0.0-git
    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.7.11.0

tfcollins, 2020-01-27:
Do you have two variants installed? PPA and standard?

The-inside-man, 2020-01-27:
If I do, I am not sure how that happened... I am new to this so it is possible I did a wrong step somewhere.

tfcollins, 2020-01-27:
I might ask the GNU Radio community. The slack channel is really helpful :)

The-inside-man, 2020-01-27:
Thank you! Just joined their chat and will see where I can get from there! Thanks again for all your help!

tfcollins, 2020-01-29:
Reported this was resolved here: https://github.com/analogdevicesinc/pyadi-iio/issues/55

closed 2020-01-29 by tfcollins