analogdevicesinc/gr-iio #63: std::mutex issue when building on ADRV9009-ZU11EG SOM

maso27, 2020-01-31:
When running "make" on the ADRV9009-ZU11EG device itself, on an image that was updated within the last week, the process errors at attr_updater_impl.cc.o.  It says 

> error: 'mutex' in namespace 'std' does not name a type

I can bypass it by editing attr_updater_impl.h, removing `#include <mutex>` and changing `std::mutex d_mtx` to `boost::mutex d_mtx`.

After that everything builds and installs fine and I can play with the demo files.

tfcollins, 2020-01-31:
What branch are you building off? There is no such block in master.

-Travis

maso27, 2020-01-31:
That explains some things.
I got here from the link on page 58 of [this presentation](https://www.gnuradio.org/grcon/grcon19/presentations/multichannel_phase_coherent_transceiver_system/Michael%20Hennerich%20-%20Multichannel%20phase%20coherent%20system.pdf).

I'm following the examples in the grcon2019 branch, using an ADRV9009-ZU11EG.

tfcollins, 2020-01-31:
O ok, yes those were never tested onboard. Only on host PC. The ARM images for the current boards are pretty ancient so mileage may vary.

closed 2020-02-26 by maso27