analogdevicesinc/gr-iio #65: Building with GNU Radio 3.8

777arc, 2020-02-07:
Are we supposed to be using master or upgrade-3.8 when building for 3.8?  When using upgrade-3.8 I get the following warning during cmake .:

-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Could NOT find PythonLibs: Found unsuitable version "2.7.17", but required is exact version "3.6" (found /usr/lib/x86_64-linux-gnu/libpython2.7.so)

and if I try to build I get the following error:

...
[ 63%] Building C object lib/CMakeFiles/gnuradio-iio.dir/lexer.c.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-iio.so....'.  Stop.

Any help would be greatly appreciated!

tfcollins, 2020-02-07:
Try installing the liborc dev packages. It seems like the ppas do not install the build dependencies for gnuradio since it didn't even install gtk on my machine either.

For the python error you will likely have to set PYTHON_EXECUTABLE. The cmake that ships with 18.04 is still kinda old and python2 vs 3 finding isn't great in my experience.

duggabe, 2020-02-08:
This morning I was successful with the following procedures:
open your text editor and add the following lines to ~/.bash_aliases :
```
export PYTHONPATH=/usr/lib/python3/dist-packages
export LD_LIBRARY_PATH=/usr/lib
```
save the file.
on your terminal enter 'exit'
start a new terminal.
sudo apt-get -y install libxml2 libxml2-dev bison flex cmake git libaio-dev libboost-all-dev swig \
libgmp-dev liborc-0.4-dev libusb-1.0-0-dev doxygen
continue with installing as in https://wiki.analog.com/resources/tools-software/linux-software/gnuradio 
except using `cmake -DCMAKE_INSTALL_PREFIX=/usr .` in place of `cmake .`

tfcollins, 2020-02-08:
Thanks for the input @duggabe

777arc, 2020-02-09:
yep, installing liborc-0.4-dev fixed it, thanks!

closed 2020-02-09 by 777arc