analogdevicesinc/gr-iio #66: Tuning parameter error using PlutoSDR

duggabe, 2020-02-08:
Using Ubuntu 18.04

gnuradio installed with
  sudo add-apt-repository ppa:mormj/gnuradio-releases
  sudo apt-get update
  sudo apt-get install gnuradio=3.8.0.0~gnuradio~bionic-8

gr-iio built with
  git clone https://github.com/analogdevicesinc/gr-iio.git
  cd ~/gr-iio
  git checkout upgrade-3.8

When running a simple PlutoSDR to FM Demod flowgraph, the initial LO Frequency is set to 99.1Mhz and I receive that station.
When I select a different frequency with a QT GUI Chooser, I get the following error:

Traceback (most recent call last):
  File "/home/barry/GRdev/Pluto_FM.py", line 121, in <lambda>
    lambda i: self.set_station(self._station_options[i]))
  File "/home/barry/GRdev/Pluto_FM.py", line 207, in set_station
    self.set_tuning(int(self.station-self.offset))
  File "/home/barry/GRdev/Pluto_FM.py", line 228, in set_tuning
    self.iio_pluto_source_0.set_params(int(self.tuning), self.samp_rate, 1000000, 32768, True, True, True, 'manual', 64, '', True)
  File "/usr/lib/python3/dist-packages/iio/iio_pluto_source_swig.py", line 174, in set_params
    return _iio_pluto_source_swig.pluto_source_sptr_set_params(self, *args, **kwargs)
TypeError: pluto_source_sptr_set_params() takes at most 11 arguments (12 given)

tfcollins, 2020-02-09:
This should be fixed with this PR: https://github.com/analogdevicesinc/gr-iio/pull/61 Just waiting on something before we can merge it.

duggabe, 2020-02-14:
I confirm the PR 61 fixed this issue.

closed 2020-02-14 by duggabe