analogdevicesinc/gr-iio #68: PlutoSDR not recognized as a hardware device

duggabe, 2020-02-14:
Ubuntu 18.04
Gnu Radio 3.8
GRC 3.8.0.0
branch: upgrade-3.8
When running a flowgraph with a PlutoSDR as the output device, I get the following warning:
Generating: '/home/barry/RTTY_transmit/RTTY_Pluto.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

It does work properly without the Throttle block.

tfcollins, 2020-02-14:
The throttle flag is set correctly: https://github.com/analogdevicesinc/gr-iio/blob/upgrade-3.8/grc/iio_pluto_source.block.yml#L3

Can you provide the flowgraph?

duggabe, 2020-02-14:
<pre>
options:
  parameters:
    author: Barry Duggan
    category: '[GRC Hier Blocks]'
    cmake_opt: ''
    comment: ''
    copyright: ''
    description: Radioteletype transmitter
    gen_cmake: 'On'
    gen_linking: dynamic
    generate_options: qt_gui
    hier_block_src_path: '.:'
    id: RTTY_Pluto
    max_nouts: '0'
    output_language: python
    placement: (0,0)
    qt_qss_theme: ''
    realtime_scheduling: ''
    run: 'True'
    run_command: '{python} -u {filename}'
    run_options: prompt
    sizing_mode: fixed
    thread_safe_setters: ''
    title: RTTY_Pluto
    window_size: ''
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [16, 12.0]
    rotation: 0
    state: enabled

blocks:
- name: baud_period
  id: variable
  parameters:
    comment: ''
    value: '0.022'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [472, 12.0]
    rotation: 0
    state: enabled
- name: center
  id: variable
  parameters:
    comment: ''
    value: '2210'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [200, 84.0]
    rotation: 0
    state: enabled
- name: fsk_deviation
  id: variable
  parameters:
    comment: ''
    value: '170'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [296, 84.0]
    rotation: 0
[... 522 lines cut ...]

connections:
- [analog_const_source_x_0, '0', blocks_sub_xx_0, '0']
- [analog_sig_source_x_0_0, '0', blocks_multiply_xx_0, '1']
- [analog_sig_source_x_0_0_0, '0', blocks_multiply_xx_1, '0']
- [blocks_add_xx_0, '0', blocks_repeat_1, '0']
- [blocks_add_xx_0, '0', qtgui_time_sink_x_0, '1']
- [blocks_float_to_complex_0, '0', iio_pluto_sink_0, '0']
- [blocks_multiply_xx_0, '0', blocks_add_xx_0, '1']
- [blocks_multiply_xx_1, '0', blocks_add_xx_0, '0']
- [blocks_repeat_0, '0', blocks_uchar_to_float_0, '0']
- [blocks_repeat_1, '0', blocks_float_to_complex_0, '0']
- [blocks_sub_xx_0, '0', blocks_multiply_xx_1, '1']
- [blocks_uchar_to_float_0, '0', low_pass_filter_0, '0']
- [blocks_vector_source_x_0, '0', blocks_null_sink_0, '0']
- [epy_block_0_0, '0', blocks_repeat_0, '0']
- [epy_block_0_0, clear_input, qtgui_edit_box_msg_0, val]
- [low_pass_filter_0, '0', blocks_multiply_xx_0, '0']
- [low_pass_filter_0, '0', blocks_sub_xx_0, '1']
- [low_pass_filter_0, '0', qtgui_time_sink_x_0, '0']
- [qtgui_edit_box_msg_0, msg, epy_block_0_0, msg_in]

metadata:
  file_format: 1
</pre>

tfcollins, 2020-02-14:
Fixed here: https://github.com/analogdevicesinc/gr-iio/commit/26fa31ecf3cd6e3ffce18ca3a64972a731550c09

duggabe, 2020-02-14:
I confirm that PR 69 fixed this issue.

closed 2020-02-14 by duggabe