analogdevicesinc/gr-iio #72: Fix GNU Radio detection

wkazubski, 2020-02-21:
In branch upgrade-3.8 (and fixes-3.8) the GNU Radio is searched twice, first time in line 86:
`find_package(Gnuradio "3.8" REQUIRED)`

and then in line 113:
`find_package(Gnuradio COMPONENTS blocks analog volk pmt filter fft)`

This triggers an error while building on openSUSE:
`CMake Error at /usr/lib64/cmake/gnuradio/FindLOG4CPP.cmake:52 (add_library):
  add_library cannot create imported target "Log4Cpp::log4cpp" because
  another target with the same name already exists.
Call Stack (most recent call first):
  /usr/share/cmake/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake:25 (find_dependency)
  CMakeLists.txt:113 (find_package)
`

Please merge both lines to one (line 86):
`find_package(Gnuradio "3.8" REQUIRED COMPONENTS blocks analog volk pmt filter fft)`

Also, the line 114:
`find_package(Gnuradio PATHS "c:/Program Files/gnuradio/lib/cmake/gnuradio" REQUIRED)`
is Windows specific and does not help in Linux or Mac.

wkazubski, 2020-02-22:
Thanks,
now GNU Radio is detected correctly and build goes on until linking:
`[ 55%] Linking CXX shared library libgnuradio-iio.so
cd /home/wk/rpmbuild/BUILD/gr-iio/build/lib && /usr/bin/cmake -E cmake_link_script CMakeFiles/gnuradio-iio.dir/link.txt --verbose=1
/usr/bin/c++ -fPIC -O2 -g -m64 -fmessage-length=0 -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -DNDEBUG -O2 -g -DNDEBUG  -Wl,--as-needed -Wl,--no-undefined -Wl,-z,now -shared -Wl,-soname,libgnuradio-iio.so... -o libgnuradio-iio.so.... CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o CMakeFiles/gnuradio-iio.dir/device_sink_impl.cc.o CMakeFiles/gnuradio-iio.dir/fmcomms2_source_impl.cc.o CMakeFiles/gnuradio-iio.dir/fmcomms2_sink_impl.cc.o CMakeFiles/gnuradio-iio.dir/fmcomms5_source_impl.cc.o CMakeFiles/gnuradio-iio.dir/fmcomms5_sink_impl.cc.o CMakeFiles/gnuradio-iio.dir/pluto_source_impl.cc.o CMakeFiles/gnuradio-iio.dir/pluto_sink_impl.cc.o CMakeFiles/gnuradio-iio.dir/attr_sink_impl.cc.o CMakeFiles/gnuradio-iio.dir/attr_source_impl.cc.o CMakeFiles/gnuradio-iio.dir/converter_ss_impl.cc.o CMakeFiles/gnuradio-iio.dir/iio_math_impl.cc.o CMakeFiles/gnuradio-iio.dir/iio_math_gen_impl.cc.o CMakeFiles/gnuradio-iio.dir/iio_power_ff_impl.cc.o CMakeFiles/gnuradio-iio.dir/iio_modulo_ff_impl.cc.o CMakeFiles/gnuradio-iio.dir/iio_modulo_const_ff_impl.cc.o CMakeFiles/gnuradio-iio.dir/lexer.c.o CMakeFiles/gnuradio-iio.dir/parser.c.o  /usr/lib64/libgnuradio-analog.so.3.8.0.0 /usr/lib64/libgnuradio-filter.so.3.8.0.0 /usr/lib64/libgnuradio-fft.so.3.8.0.0 /usr/lib64/libiio.so /usr/lib64/libad9361.so /usr/lib64/libgnuradio-blocks.so.3.8.0.0 /usr/lib64/libgnuradio-runtime.so.3.8.0.0 /usr/lib64/libgnuradio-pmt.so.3.8.0.0 /usr/lib64/libboost_program_options.so.1.71.0 /usr/lib64/libboost_filesystem.so.1.71.0 /usr/lib64/libboost_system.so.1.71.0 /usr/lib64/libboost_regex.so.1.71.0 /usr/lib64/libboost_thread.so.1.71.0 -lpthread /usr/lib64/liblog4cpp.so /usr/lib64/libgmpxx.so /usr/lib64/libgmp.so -lrt /usr/lib64/libfftw3f.so /usr/lib64/libfftw3f_threads.so /usr/lib64/libvolk.so.2.1 -ldl /usr/lib64/liborc-0.4.so -lm 
/usr/lib64/gcc/x86_64-suse-linux/9/../../../../x86_64-suse-linux/bin/ld: CMakeFiles/gnuradio-iio.dir/attr_source_impl.cc.o: in function `bool boost::condition_variable::wait_for<long, boost::ratio<1l, 1000l>, bool (*)()>(boost::unique_lock<boost::mutex>&, boost::chrono::duration<long, boost::ratio<1l, 1000l> > const&, bool (*)())':
/usr/include/boost/thread/pthread/condition_variable_fwd.hpp:321: undefined reference to `boost::chrono::steady_clock::now()'
collect2: error: ld returned 1 exit status
make[2]: *** [lib/CMakeFiles/gnuradio-iio.dir/build.make:373: lib/libgnuradio-iio.so....] Error 1
`

tfcollins, 2020-02-25:
Do you have boost chrono?

wkazubski, 2020-02-26:
Yes. It was installed at that  time, but is neither checked for by cmake nor used for linking. In openSUSE, gnuradio 3.8 checks for those boost modules: date_time program_options filesystem system regex thread unit_test_framework.

tfcollins, 2020-02-26:
Let me load up a SUSE vm and check this out

commodo, 2020-02-26:
Try docker.
It’s faster.
There should be some prebuilt images on dockerhub.

commit 12f90408 (2020-02-24): Update cmake to handle gr3.8 and gr3.9

Due to the existence of the ubuntu ppa's, development builds of gnuradio are becoming more common with the version 3.9. This updates cmake to handle 3.8 and 3.9 with some minor changes for windows includes. Fixes #72
commit 9fef00cb (2020-02-25): Update cmake to handle gr3.8 and gr3.9

Due to the existence of the ubuntu ppa's, development builds of gnuradio are becoming more common with the version 3.9. This updates cmake to handle 3.8 and 3.9 with some minor changes for windows includes. Fixes #72
PR #73 (2020-02-25): Update cmake to handle gr3.8 and gr3.9
Due to the existence of the ubuntu ppa's, development builds of gnuradio are becoming more common with the version 3.9. This updates cmake to handle 3.8 and 3.9 with some minor changes for windows includes. Fixes #72