analogdevicesinc/gr-iio #74: 3.9 Support

anilgurses, 2020-03-22:
Hi,
I tried to build gr-iio for GNURadio 3.9. However, some oft he package names has changed with version 3.9. CMake can't locate the gr-analog libs. I looked for a branch and I didn't find anything that related to it. Has anyone succeed on building gr-iio for GNURadio 3.9?

tfcollins, 2020-03-22:
The upgrade-3.8 branch should handle 3.9.

Can you provide more information about your setup and what the exact error is?

anilgurses, 2020-03-24:
I installed GnuRadio 3.9 on ubuntu 18.04. I am able to compile libiio and libad9361 without any error. However, gr-iio fails on make command.  It can't locate gr-analog libs. I uninstalled all the packages and blocks and re-installed GnuRadio 3.9. It hasn't solved the problem. I also check the branch to upgrade 3.8 and it didn't work too. I downgraded GnuRadio to 3.7 now and it has no problem.  I will install the GnuRadio 3.9 in a day or two and I will send you the log output.

anilgurses, 2020-03-25:
The error on master branch;
```
fatal error: gnuradio/analog/sig_source_f.h: No such file or directory
 #include <gnuradio/analog/sig_source_f.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```
The error on upgrade 3.8 branch;

```
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-iio.so....'.  Stop.
```

anilgurses, 2020-03-25:
I solved it with reinstalling the liborc-0.4-dev. I built with upgrade 3.8 branch. Thanks

anilgurses, 2020-03-25:
I solved it with reinstalling the liborc-0.4-dev. I built with upgrade 3.8 branch. However, GnuRadio doesn't show the modules and iio module can't imported.

tfcollins, 2020-03-25:
What error do you get on the import?

anilgurses, 2020-03-25:
```
>>> import iio
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'iio'
```

tfcollins, 2020-03-25:
Is iio on your pythonpath?

anilgurses, 2020-03-25:
Yes. I controlled it twice and the as wiki said that I copied the package

`cp -r /usr/local/lib/python3/dist-packages/iio /usr/lib/python3/dist-packages/gnuradio/`

tfcollins, 2020-03-25:
What is the output of `python3 -c "import os;print(os.sys.path)"`

anilgurses, 2020-03-25:
Output is 

`['', '/usr/lib/python36.zip', '/usr/lib/python3.6', '/usr/lib/python3.6/lib-dynload', '/usr/local/lib/python3.6/dist-packages', '/usr/lib/python3/dist-packages']
`

tfcollins, 2020-03-25:
Based on that and how you moved the python files you would have to do: `from gnuradio import iio`.

The wiki describes the install for 3.7.  3.8+ puts things in a slightly different place (notice the gnuradio not in the original path).

anilgurses, 2020-03-25:
iio is now imported. Thank you for that. What about the blocks ui. I can't see them right now.

anilgurses, 2020-03-25:
iio is now imported. Thank you for that. What about the xml blocks? I can't see them again. Is there another thing to do while I am building gr-iio.

anilgurses, 2020-03-25:
I also built the cmake-update-3.8 branch. It successfully built too. However, grc blocks aren't at GnuRadio interface

tfcollins, 2020-03-25:
I would not put iio in the gnuradio subfolder, `/usr/lib/python3/dist-packages` is correct.

In regards to the GRC files, I assume you mean yml and not xml files. xml are only supported in GR <=3.7. CMake will put them in the correct place unless you have a non-standard xml path set. There was a similar question here: https://github.com/analogdevicesinc/pyadi-iio/issues/55

anilgurses, 2020-03-25:
Thank you for your help. Now, I am able to see those blocks. I copied all yml files from `/usr/share/gr-iio/grc/blocks` to `/usr/share/gnuradio/grc/block` and it's solved the problem

closed 2020-03-25 by anilgurses

japm48, 2020-03-25:
> I would not put iio in the gnuradio subfolder, `/usr/lib/python3/dist-packages` is correct.

But then it conflicts with [iio python bindings](https://github.com/analogdevicesinc/libiio/tree/master/bindings/python) (same package name: `iio` instead of `gnuradio.iio`), which is a dependency of [pyadi-iio](https://github.com/analogdevicesinc/pyadi-iio/).

japm48, 2020-03-25:
@tfcollins
A possible solution is to rename renaming this OOT module as `griio` (so that the out of tree policy is respected and, at the same time, avoiding the conflict at the same time).

But, as apparently this module is going [to be merged](https://github.com/gnuradio/gnuradio/pull/2840) in the future, I would suggest sticking with `gnuradio.iio`.

tfcollins, 2020-03-25:
@japm48 At this point no. It would break a lot of things.

We might go back to putting things in the gnuradio subfolder again. Ideally our upstream PR gets merged and you will just get gr-iio with gnuradio and won't after to worry about this :)

japm48, 2020-03-25:
> @japm48 At this point no. It would break a lot of things.

Fair enough.

> We might go back to putting things in the gnuradio subfolder again. Ideally our upstream PR gets merged and you will just get gr-iio with gnuradio and won't after to worry about this :)

Nice. Thanks for your work.