analogdevicesinc/gr-iio #78: USRP E310 and gr-iio

krono86, 2020-04-08:
Hello,
I'm trying to use GNU Radio and USRP E310 in "network mode" and I found out that this is possible using iio drivers and gr-iio.
To simplify the interface I added source/sink blocks in gr-iio module, adapting fmcomms2 ones.
I would like share with you these blocks, they could be useful to someone.

Now, I have some questions:
- E310 blocks the receiveing rf path selection. The right selection, base on the selected frequencies, is made by the driver (iio), or in-hardware?
- How can I access to the switch between TRX and RX channels on the receiving path via iio driver? I need to implement a TDD communication, using the same antenna.
Thank you.

Ivan

tfcollins, 2020-04-08:
How are these different than the existing FMComms blocks? They seem just have a name change from FMComms->E310.

1. Driver changes frequencies on all platforms that use AD936x.
2. I don't believe this is possible through a standard attribute right now. Essentially you need to change the GPIO settings (adi,gpio-settings) in the device tree, which is possible through debug fs attributes and re-initializing the driver. The driver handles the filter bank

krono86, 2020-04-08:
The block are almost the same for fmcomms2.
On E310 it‘s not possible to select the rf paths, thus I removed such a feature in both, source and sink code. Furthermore, I removed the exceptions provided for fmcomms4 (ad9364).
I also found out, and corrected in E310 block, a bug in the yml of the fmcomms2 source: gain box was not revealed when manual mode was selected.

1) I was not talking about frequency set, but rf path select.
2) Ok, I will investigate on the driver side.

Ivan

tfcollins, 2020-04-08:
1. Driver handles this.