analogdevicesinc/gr-iio #79: Undefined symbol in libgnuradio-iio: _ZN5boost6chrono12steady_clock3nowEv

ivanmikh, 2020-04-28:
I have a problem with FMComms2/3/4 or PlutoSDR blocks in GNU Radio.
I have GNU Radio 3.8 and PlutoSDR connected via USB.

I am trying to run cyclic-sine_pluto.grc from gr-iio/iio-examples and I get an error:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3/dist-packages/iio/__init__.py", line 30, in <module>
    from .iio_swig import *
  File "/usr/local/lib/python3/dist-packages/iio/iio_swig.py", line 13, in <module>
    from . import _iio_swig
ImportError: /usr/local/lib/x86_64-linux-gnu/libgnuradio-iio.so...: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ivan/Documents/tools/analog_devices/gr-iio/iio-examples/top_block.py", line 34, in <module>
    import iio
  File "/usr/local/lib/python3/dist-packages/iio/__init__.py", line 36, in <module>
    from .iio_swig import *
  File "/usr/local/lib/python3/dist-packages/iio/iio_swig.py", line 13, in <module>
    from . import _iio_swig
ImportError: /usr/local/lib/x86_64-linux-gnu/libgnuradio-iio.so...: undefined symbol: _ZN5boost6chrono12steady_clock3nowEv
```
Symbol demangling:
```
ivan@lubuntu20vm:~$c++filt _ZN5boost6chrono12steady_clock3nowEv
boost::chrono::steady_clock::now()
```

Also seems like the symbol is undefined in library:
```
ivan@lubuntu20vm:~$ nm /usr/local/lib/x86_64-linux-gnu/libgnuradio-iio.so.... | grep _ZN5boost6chrono12steady_clock3nowEv
                 U _ZN5boost6chrono12steady_clock3nowEv
```

I have compiled and installed library using [AD wiki](https://wiki.analog.com/resources/tools-software/linux-software/gnuradio#download_and_build_libiio).

Does anybody know how to fix this issue? I would be really grateful for any help or tip.

ivanmikh, 2020-04-28:
So the problem was that library was not linking against libboost_timer

I was able to fix it by adding this to CMakeLists.txt:
```
add_definitions(-lboost_chrono)
add_definitions(-DBOOST_CHRONO_DYN_LINK)
add_definitions(-lboost_timer)
add_definitions(-DBOOST_TIMER_DYN_LINK)
```
And adding -lboost_chrono -lboost_timer flags to gr-iio/build/lib/CMakeFiles/gnuradio-iio.dir/link.txt.

This is just a temporary hack, what would be the right way to fix this issue?

adisuciu, 2020-04-30:
Can you tell us what version of boost are you running and what version of gnuradio (and where you installed from ) ?

ivanmikh, 2020-05-03:
I installed GNU Radio from Ubuntu repository using apt. My OS is Lubuntu 20.04 LTS (Focal Fossa).

GNU radio:
```
ii  gnuradio                                      3.8.1.0~rc1-2build2                 amd64        GNU Radio Software Radio Toolkit
ii  gnuradio-dev:amd64                            3.8.1.0~rc1-2build2                 amd64        GNU Software Defined Radio toolkit development
```

libbosst:
```
ii  libboost-all-dev                              1.71.0.0ubuntu2                     amd64        Boost C++ Libraries development files (ALL) (default version)
```

Doom4535, 2020-05-07:
I can confirm this issue using Ubuntu-20.04 and that @ivanmikh's solution works.

# My Setup
* gnuradio installed from source on branch `main-3.8` with volk on `v2.2.1`
* gr-iio on branch `upgrade-3.8`
* ii_info version: 0.19
* libbost-dev -> 1.71.0.0.ubuntu2

# Extended replication
## Sources
* libini: https://github.com/pcercuei/libini/commit/3ab1d1f8836e53b0f2e8100ea5d6f1b98aa3bddf
* libiio: https://github.com/analogdevicesinc/libiio/commit/cc55f51b85c8dedeb59d10c745c7c396476ca28f
* libad9361-iio: https://github.com/analogdevicesinc/libad9361-iio/commit/c5ba32be26382a6e69db4418381d44be965c2dd7
* gnuradio: https://github.com/gnuradio/gnuradio/commit/dc3046459dc9fb2b4ccd1d1695699e6787faeb7c
  - volk: https://github.com/gnuradio/volk/commit/aae9fed1cd15f0f1695d5adad04c65cf8bbd3cc0
* gr-iio: https://github.com/analogdevicesinc/gr-iio/commit/456589948229cc7bbed4c5724811ca24116711fe
## Install Procedure
### Build and Install Dependencies
1. libini
2. libiio
3. libad9361-iio
4. gnuradio
### Patch and Install gr-iio
```{.bash}
mkdir build && cd build && cmake .. -DCMAKE_INSTALL_PREFIX=/usr
echo "
add_definitions(-lboost_chrono)
add_definitions(-DBOOST_CHRONO_DYN_LINK)
add_definitions(-lboost_timer)
add_definitions(-DBOOST_TIMER_DYN_LINK)
"
sed -i.bak "$ s/$/ -lboost_chrono -lboost_timer/" lib/CMakeFiles/gnuradio-iio.dir/link.txt
make -j $(nproc)
sudo make install
sudo ldconfig
```

closed 2020-06-03 by tfcollins