analogdevicesinc/gr-iio #80: Using two USRP E310 at the same time

krono86, 2020-05-13:
Hello,
I'm trying to use two USRP E310 at the same time connecting via ethernet my PC and both the E310 to a network switch.
I found out GNU Radio is not able to communicate with the two devices at the same time! I obtain the message:

```
ERROR: Unable to find host: Name or service not known
...
..., line 125, in __init__
    self.iio_e310_source_0 = iio.e310_source_f32c('ip:192.168.10.1', carrier_freq, samp_rate, BW, True, False, 32768, True, True, True, 'manual', 0, 'slow_attack', 64, '', True)
  File "/usr/local/lib/python2.7/dist-packages/iio/iio_swig.py", line 666, in make
    return _iio_swig.e310_source_f32c_make(*args, **kwargs)
RuntimeError: Unable to create context
```

What's wrong?
Thank you.

Ivan

tfcollins, 2020-05-13:
Are both E310's getting valid IPs from your router?

-Travis

krono86, 2020-05-13:
Yes, and I can ping both from PC.

Ivan

tfcollins, 2020-05-13:
Can you SSH to both machines? Can you get a list of devices with iio_attr (`iio_attr -u ip:<address> -d>`?

-Travis

krono86, 2020-05-13:
Yes, I can SSH to both machine. I now noticed when I'm connected with both using SSH, after interacting with one I have to wait some seconds to interact with the other one.
It's so strange...

Ivan

tfcollins, 2020-05-13:
It sounds like the IPs aren't set correctly.

-Travis

krono86, 2020-05-13:
Solved!
It was not an IP problem, but I found out that the two USRP had the same MAC address!
I assigned the right MAC addresses in uenv.txt and now they work like a charm.
Thank you for support!

Ivan

closed 2020-05-13 by krono86