analogdevicesinc/gr-iio #81: Burst transmission

krono86, 2020-05-22:
Hello,
I know it is possible to use iio devices in burst mode setting the kernel buffer to 1.
Is there a way to use gr-iio sink blocks in burst mode?
Thank you.

Ivan

tfcollins, 2020-05-22:
You will have to modify the source code to change the number of kernel buffers.

Why do you want to change the number of buffers? GNU Radio isn't really great at handling discontinuity in data streams which setting kernel buffers to 1 would do.

krono86, 2020-05-25:
I red this solution here:
https://ez.analog.com/adieducation/university-program/f/q-a/105399/adalm-pluto-burst-mode/314111#314111
It's the operation done by Matlab in burst mode.
I hope this will fix my errors in the burst generation.
Thank you.

Ivan

tfcollins, 2020-05-26:
Can you explain what you are trying to do? GNU Radio and MATLAB have very different paradigms.

-Travis

krono86, 2020-05-26:
Using ZMQ I'm sending messages to my flowgraph every 0.5 s, each message is coded onto a burst (packet) of length 120 us that has to be transmitted by the USRP.
During the time between two packets I want transmit nothing.
Is this the righy way to obtain a consistent transmission of packets?
Thank you.

Ivan

tfcollins, 2020-05-27:
The sink block operates on buffers. So when it has a full buffer a data it well send that from the host to the radio, and then start queue buffers behind it. Burst mode would make this worse since it would prevent you from queuing buffers since you can only have 1.

Can you explain more what you mean by consistent transmissions and what you are observing?

-Travis