analogdevicesinc/gr-iio #82: Compiling for GNU Radio 3.8

rkristijanto, 2020-05-31:
Hello,

Can you please help me with the compilation for GNU Radio 3.8. 
I have this error below. Thank you for your help

`(base) ronald@ronald-laptop:~/Documents/Mentahan/gr-iio$ cmake .
-- Build type not specified: defaulting to release.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.2") 
-- Extracting version information from git describe...
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.2") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ronald/Documents/Mentahan/gr-iio
`
`(base) ronald@ronald-laptop:~/Documents/Mentahan/gr-iio$ make
[  2%] [BISON][parser] Building parser with bison 3.5.1
[  5%] [FLEX][lexer] Building scanner with flex 2.6.4
Scanning dependencies of target gnuradio-iio
[  7%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o
[ 10%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/device_sink_impl.cc.o
[ 13%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/fmcomms2_source_impl.cc.o
[ 15%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/fmcomms2_sink_impl.cc.o
[ 18%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/fmcomms5_source_impl.cc.o
[ 21%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/fmcomms5_sink_impl.cc.o
[ 23%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/pluto_source_impl.cc.o
[ 26%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/pluto_sink_impl.cc.o
[ 28%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/attr_sink_impl.cc.o
[ 31%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/attr_source_impl.cc.o
[ 34%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/converter_ss_impl.cc.o
[ 36%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/iio_math_impl.cc.o
[ 39%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/iio_math_gen_impl.cc.o
[ 42%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/iio_power_ff_impl.cc.o
In file included from /usr/local/include/volk/volk.h:31:0,
                 from /home/ronald/Documents/Mentahan/gr-iio/lib/iio_power_ff_impl.cc:29:
/usr/local/include/volk/volk_common.h: In function 'float log2f_non_ieee(float)':
/usr/local/include/volk/volk_common.h:153:24: error: 'isinf' was not declared in this scope
     return isinf(result) ? copysignf(127.0f, result) : result;
                        ^
/usr/local/include/volk/volk_common.h:153:24: note: suggested alternative:
In file included from /home/linuxbrew/.linuxbrew/Cellar/gcc/5.5.0_7/include/c++/5.5.0/complex:44:0,
                 from /usr/include/gnuradio/gr_complex.h:26,
                 from /usr/include/gnuradio/types.h:31,
                 from /usr/include/gnuradio/runtime_types.h:27,
                 from /usr/include/gnuradio/io_signature.h:27,
                 from /usr/include/gnuradio/basic_block.h:27,
                 from /usr/include/gnuradio/block.h:27,
                 from /usr/include/gnuradio/sync_block.h:27,
                 from /home/ronald/Documents/Mentahan/gr-iio/lib/../include/iio/power_ff.h:27,
                 from /home/ronald/Documents/Mentahan/gr-iio/lib/iio_power_ff_impl.h:26,
                 from /home/ronald/Documents/Mentahan/gr-iio/lib/iio_power_ff_impl.cc:26:
/home/linuxbrew/.linuxbrew/Cellar/gcc/5.5.0_7/include/c++/5.5.0/cmath:626:5: note:   'std::isinf'
     isinf(_Tp __x)
     ^
make[2]: *** [lib/CMakeFiles/gnuradio-iio.dir/build.make:243: lib/CMakeFiles/gnuradio-iio.dir/iio_power_ff_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:219: lib/CMakeFiles/gnuradio-iio.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
`

Thank you once again

tfcollins, 2020-06-01:
You seem to be missing volk which is a gnuradio dependency. How did you install gnuradio?

-Travis

rkristijanto, 2020-06-02:
I installed gnuradio from the Ubuntu repository 
[image]

and yes I think I installed volk by compiling it from github 

https://github.com/gnuradio/volk

```
(base) ronald@ronald-laptop:~/Documents/Mentahan/volk-master$ ls -la
total 292
drwxrwxr-x 15 ronald ronald   4096 May 19 12:01 .
drwxrwxr-x 15 ronald ronald   4096 May 31 13:57 ..
drwxrwxr-x  2 ronald ronald   4096 May 12 22:11 apps
-rw-rw-r--  1 ronald ronald   4457 May 12 22:11 appveyor.yml
drwxrwxr-x  9 ronald ronald   4096 May 19 12:04 build
-rw-rw-r--  1 ronald ronald  11086 May 12 22:11 CHANGELOG.md
-rw-rw-r--  1 ronald ronald   2946 May 12 22:11 .clang-format
drwxrwxr-x  5 ronald ronald   4096 May 12 22:11 cmake
-rw-rw-r--  1 ronald ronald  13823 May 12 22:11 CMakeLists.txt
-rw-rw-r--  1 ronald ronald   3495 May 12 22:11 CODE_OF_CONDUCT.md
-rw-rw-r--  1 ronald ronald  35147 May 12 22:11 COPYING
drwxrwxr-x  2 ronald ronald   4096 May 12 22:11 docs
-rw-rw-r--  1 ronald ronald 102566 May 12 22:11 Doxyfile.in
-rw-rw-r--  1 ronald ronald   6104 May 12 22:11 DoxygenLayout.xml
drwxrwxr-x  3 ronald ronald   4096 May 19 12:01 gen
drwxrwxr-x  3 ronald ronald   4096 May 12 22:11 .github
-rw-rw-r--  1 ronald ronald     34 May 12 22:11 .gitignore
-rw-rw-r--  1 ronald ronald    961 May 12 22:11 .gitlab-ci.yml
drwxrwxr-x  3 ronald ronald   4096 May 12 22:11 include
drwxrwxr-x  3 ronald ronald   4096 May 12 22:11 kernels
-rw-rw-r--  1 ronald ronald      7 May 12 22:11 .lastrelease
drwxrwxr-x  2 ronald ronald   4096 May 12 22:11 lib
drwxrwxr-x  3 ronald ronald   4096 May 12 22:11 python
-rw-rw-r--  1 ronald ronald   2991 May 12 22:11 README.md
drwxrwxr-x  3 ronald ronald   4096 May 12 22:11 scripts
drwxrwxr-x  2 ronald ronald   4096 May 12 22:11 tmpl
drwxrwxr-x  2 ronald ronald   4096 May 12 22:11 tools
-rw-rw-r--  1 ronald ronald   3688 May 12 22:11 .travis.yml
-rw-rw-r--  1 ronald ronald  14678 May 12 22:11 versioning.md
-rw-rw-r--  1 ronald ronald    268 May 12 22:11 volk.pc.in
```

```
(base) ronald@ronald-laptop:/usr/local/include/volk$ pwd
/usr/local/include/volk
(base) ronald@ronald-laptop:/usr/local/include/volk$ ls 
asm                                                volk_32f_s32f_32f_fm_detect_32f.h
constants.h                                        volk_32f_s32f_add_32f.h
saturation_arithmetic.h                            volk_32f_s32f_calc_spectral_noise_floor_32f.h
volk_16i_32fc_dot_prod_32fc.h                      volk_32f_s32f_convert_16i.h
volk_16i_branch_4_state_8.h                        volk_32f_s32f_convert_32i.h
volk_16ic_convert_32fc.h                           volk_32f_s32f_convert_8i.h
volk_16ic_deinterleave_16i_x2.h                    volk_32f_s32f_mod_rangepuppet_32f.h
volk_16ic_deinterleave_real_16i.h                  volk_32f_s32f_multiply_32f.h
volk_16ic_deinterleave_real_8i.h                   volk_32f_s32f_normalize.h
volk_16ic_magnitude_16i.h                          volk_32f_s32f_power_32f.h
volk_16i_convert_8i.h                              volk_32f_s32f_s32f_mod_range_32f.h
volk_16ic_s32f_deinterleave_32f_x2.h               volk_32f_s32f_stddev_32f.h
volk_16ic_s32f_deinterleave_real_32f.h             volk_32f_sin_32f.h
volk_16ic_s32f_magnitude_32f.h                     volk_32f_sqrt_32f.h
volk_16ic_x2_dot_prod_16ic.h                       volk_32f_stddev_and_mean_32f_x2.h
volk_16ic_x2_multiply_16ic.h                       volk_32f_tan_32f.h
volk_16i_max_star_16i.h                            volk_32f_tanh_32f.h
volk_16i_max_star_horizontal_16i.h                 volk_32f_x2_add_32f.h
volk_16i_permute_and_scalar_add.h                  volk_32f_x2_divide_32f.h
volk_16i_s32f_convert_32f.h                        volk_32f_x2_dot_prod_16i.h
volk_16i_x4_quad_max_star_16i.h                    volk_32f_x2_dot_prod_32f.h
volk_16i_x5_add_quad_16i_x4.h                      volk_32f_x2_fm_detectpuppet_32f.h
volk_16u_byteswap.h                                volk_32f_x2_interleave_32fc.h
volk_16u_byteswappuppet_16u.h                      volk_32f_x2_max_32f.h
[... 28 lines cut ...]
volk_32fc_s32fc_rotatorpuppet_32fc.h               volk_8ic_s32f_deinterleave_real_32f.h
volk_32fc_s32fc_x2_rotator_32fc.h                  volk_8ic_x2_multiply_conjugate_16ic.h
volk_32fc_s32f_deinterleave_real_16i.h             volk_8ic_x2_s32f_multiply_conjugate_32fc.h
volk_32fc_s32f_magnitude_16i.h                     volk_8i_s32f_convert_32f.h
volk_32fc_s32f_power_32fc.h                        volk_8u_conv_k7_r2puppet_8u.h
volk_32fc_s32f_power_spectral_densitypuppet_32f.h  volk_8u_x2_encodeframepolar_8u.h
volk_32fc_s32f_power_spectrum_32f.h                volk_8u_x3_encodepolar_8u_x2.h
volk_32fc_s32f_x2_power_spectral_density_32f.h     volk_8u_x3_encodepolarpuppet_8u.h
volk_32fc_x2_add_32fc.h                            volk_8u_x4_conv_k7_r2_8u.h
volk_32fc_x2_conjugate_dot_prod_32fc.h             volk_alloc.hh
volk_32fc_x2_divide_32fc.h                         volk_avx2_intrinsics.h
volk_32fc_x2_dot_prod_32fc.h                       volk_avx_intrinsics.h
volk_32fc_x2_multiply_32fc.h                       volk_common.h
volk_32fc_x2_multiply_conjugate_32fc.h             volk_complex.h
volk_32fc_x2_s32fc_multiply_conjugate_add_32fc.h   volk_config_fixed.h
volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h    volk_cpu.h
volk_32fc_x2_square_dist_32f.h                     volk.h
volk_32f_exp_32f.h                                 volk_malloc.h
volk_32f_expfast_32f.h                             volk_neon_intrinsics.h
volk_32f_index_max_16u.h                           volk_prefs.h
volk_32f_index_max_32u.h                           volk_sse3_intrinsics.h
volk_32f_invsqrt_32f.h                             volk_sse_intrinsics.h
volk_32f_log2_32f.h                                volk_typedefs.h
volk_32f_null_32f.h                                volk_version.h
```

tfcollins, 2020-06-02:
GNU Radio requires specific versions of volk and building it separately might not be a great idea.
It should be installed as part of the development package (gnuradio-dev) or from within the GNU Radio source tree with it.

Try `apt install gnuradio-dev`, then redownload and build gr-iio.

-Travis

rkristijanto, 2020-06-02:
Wow wow wow ... you are so right ! ... Thank you so much ... when I do `apt install gnuradio-dev` I already have that. So I just uninstall my 'other volk' ( the one that I got from github ) ... I just `sudo make uninstall `... and .... here it is ... apparently I already have volk anyway ... :) Super awesome ... 

Thank you thank you thank you so much for your help Travis

I am ready to rock now ... Super awesome :+1:  :1st_place_medal: 
Now they are appearing in my GNU Radio Companion :+1: 

[image]

[image]

-- Ronald --

```
ronald@ronald-laptop:~/Documents/Mentahan/gr-iio$ cmake .
-- Build type not specified: defaulting to release.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.2") 
-- Extracting version information from git describe...
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.2") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ronald/Documents/Mentahan/gr-iio

ronald@ronald-laptop:~/Documents/Mentahan/gr-iio$ make
Scanning dependencies of target gnuradio-iio
[  2%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/iio_power_ff_impl.cc.o
[  5%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/iio_modulo_ff_impl.cc.o
[  7%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/iio_modulo_const_ff_impl.cc.o
[ 10%] Building C object lib/CMakeFiles/gnuradio-iio.dir/lexer.c.o
[ 13%] Building C object lib/CMakeFiles/gnuradio-iio.dir/parser.c.o
[ 15%] Linking CXX shared library libgnuradio-iio.so
[ 55%] Built target gnuradio-iio
Scanning dependencies of target iio_pluto_sink_swig_swig_compilation
[ 57%] Swig source iio_pluto_sink_swig.i
Deprecated command line option: -modern. This option is now always on.
[ 57%] Built target iio_pluto_sink_swig_swig_compilation
Scanning dependencies of target iio_pluto_sink_swig
[ 60%] Building CXX object swig/CMakeFiles/iio_pluto_sink_swig.dir/CMakeFiles/iio_pluto_sink_swig.dir/iio_pluto_sink_swigPYTHON_wrap.cxx.o
[ 63%] Linking CXX shared module _iio_pluto_sink_swig.so
[ 63%] Built target iio_pluto_sink_swig
Scanning dependencies of target iio_swig_swig_compilation
[ 65%] Swig source iio_swig.i
Deprecated command line option: -modern. This option is now always on.
[ 65%] Built target iio_swig_swig_compilation
Scanning dependencies of target iio_swig
[ 68%] Building CXX object swig/CMakeFiles/iio_swig.dir/CMakeFiles/iio_swig.dir/iio_swigPYTHON_wrap.cxx.o
[ 71%] Linking CXX shared module _iio_swig.so
[ 71%] Built target iio_swig
Scanning dependencies of target iio_pluto_source_swig_swig_compilation
[ 73%] Swig source iio_pluto_source_swig.i
Deprecated command line option: -modern. This option is now always on.
[ 73%] Built target iio_pluto_source_swig_swig_compilation
Scanning dependencies of target iio_pluto_source_swig
[ 76%] Building CXX object swig/CMakeFiles/iio_pluto_source_swig.dir/CMakeFiles/iio_pluto_source_swig.dir/iio_pluto_source_swigPYTHON_wrap.cxx.o
[ 78%] Linking CXX shared module _iio_pluto_source_swig.so
[ 78%] Built target iio_pluto_source_swig
Scanning dependencies of target pygen_swig_08010
[ 81%] Generating iio_pluto_source_swig.pyc
[ 84%] Generating iio_pluto_source_swig.pyo
[ 84%] Built target pygen_swig_08010
Scanning dependencies of target pygen_swig_7a174
[ 86%] Generating iio_swig.pyc
[ 89%] Generating iio_swig.pyo
[... 62 lines cut ...]
-- Installing: /usr/local/lib/python3/dist-packages/iio/iio_pluto_sink_swig.py
-- Installing: /usr/local/lib/python3/dist-packages/iio/iio_pluto_sink_swig.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iio/iio_pluto_sink_swig.pyo
-- Installing: /usr/local/include/iio/iio/swig/iio_swig.i
-- Installing: /usr/local/include/iio/iio/swig/iio_pluto_source_swig.i
-- Installing: /usr/local/include/iio/iio/swig/iio_pluto_sink_swig.i
-- Installing: /usr/local/lib/python3/dist-packages/iio/__init__.py
-- Installing: /usr/local/lib/python3/dist-packages/iio/attr_updater.py
-- Installing: /usr/local/lib/python3/dist-packages/iio/__init__.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iio/attr_updater.pyc
-- Installing: /usr/local/lib/python3/dist-packages/iio/__init__.pyo
-- Installing: /usr/local/lib/python3/dist-packages/iio/attr_updater.pyo
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio.tree.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_attr_sink.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_attr_source.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_attr_updater.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_device_sink.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_device_source.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_fmcomms2_sink.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_fmcomms2_source.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_fmcomms5_sink.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_fmcomms5_source.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_pluto_sink.block.yml
-- Installing: /usr/local/share/gnuradio/grc/blocks/iio_pluto_source.block.yml
```

closed 2020-06-02 by rkristijanto