analogdevicesinc/gr-iio #83: Cannot build gr-iio on Ubuntu 16.04

clouca, 2020-06-01:
I am trying to compile gr-iio in Ubuntu 16.04 Xenial with Gnuradio 3.7.11 already install and get the same error every time 
/device_source_impl.cc:149:52: error: ‘iio_create_context_from_uri’ was not declared in this scope
       ctx = iio_create_context_from_uri(uri.c_str());
Reviewing the source code I cannot find anywhere this function except device_source_impl.cc so it is obvious that will cause error .
Is there any fix ? 
All requirements installed using apt except from gnuradio

tfcollins, 2020-06-01:
What version of libiio do you have installed?

-Travis

clouca, 2020-06-10:
Hello,
I have managed to solve the issue. In order to install gr-iio in Ubuntu 16.04 Xenial, do not install libiio and libad9361-iio from apt package manager. They need to be installed from source code also. It seems that Ubuntu 16.04 is not up to date according to these libraries

closed 2020-06-10 by clouca