analogdevicesinc/gr-iio #84: python3 unsupported?

ZeroChaos-, 2020-06-01:
Here's my build log.  Python 3.7 is found by cmake but the build fails later when it hits some obviously python2 code.

Is there a branch I should be using for gr3.8 and python3 or?
```
>>> Jobs: 0 of 1 complete, 1 failed                 Load avg: 4.08, 2.68, 3.49
 * Package:    net-wireless/gr-iio-0.3_p20191029
 * Repository: maintainer-zero
 * Maintainer: zerochaos@gentoo.org radio@gentoo.org
 * USE:        abi_x86_64 amd64 elibc_glibc kernel_linux python_single_target_python3_7 userland_GNU
 * FEATURES:   compressdebug network-sandbox preserve-libs sandbox splitdebug userpriv usersandbox
 * Using python3.7 to build
>>> Unpacking source...
>>> Unpacking gr-iio-0.3_p20191029.tar.gz to /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work
>>> Source unpacked in /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work
>>> Preparing source in /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338 ...
 * Hardcoded definition(s) removed in CMakeLists.txt:
 *     set(CMAKE_BUILD_TYPE "Release")
 *  set(CMAKE_BUILD_TYPE ${CMAKE_BUILD_TYPE} CACHE STRING "")
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338 ...
>>> Working in BUILD_DIR: "/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-0.3_p20191029_build"
cmake -C /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-0.3_p20191029_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Gentoo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-0.3_p20191029_build/gentoo_toolchain.cmake  /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338
loading initial cache file /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-0.3_p20191029_build/gentoo_common_config.cmake
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found suitable version "1.72.0", minimum required is "1.35") found components: filesystem system thread date_time chrono atomic 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen dot 
-- Found LOG4CPP: /usr/lib/liblog4cpp.so
-- Found PkgConfig: x86_64-pc-linux-gnu-pkg-config (found version "0.29.2") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.0
-- Found GMP: /usr/lib64/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib64/libgmpxx.so  
-- Found Boost: /usr/include (found suitable version "1.72.0", minimum required is "1.72.0") found components: date_time program_options filesystem system regex thread chrono atomic 
-- Found VOLK: Volk::volk  
CMake Warning (dev) at /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake:48 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  CMakeLists.txt:123 (find_package)
[... 87 lines cut ...]

Traceback (most recent call last):
  File "/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-0.3_p20191029_build/python_compile_helper.py", line 6, in <module>
    py_compile.compile(file=src, cfile=gen, doraise=True)
  File "/usr/lib/python3.7/py_compile.py", line 147, in compile
    raise py_exc
py_compile.PyCompileError:   File "/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/python/iio/attr_updater.py", line 74
    print "Error: Failed to publish message"
                                           ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print("Error: Failed to publish message")?

[5/24] /usr/bin/x86_64-pc-linux-gnu-g++ -DGR_VERSION_3_7_OR_LESS -Dgnuradio_iio_EXPORTS -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/include -Ilib -Iinclude  -DNDEBUG -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fPIC   -fvisibility=hidden -MD -MT lib/CMakeFiles/gnuradio-iio.dir/pluto_source_impl.cc.o -MF lib/CMakeFiles/gnuradio-iio.dir/pluto_source_impl.cc.o.d -o lib/CMakeFiles/gnuradio-iio.dir/pluto_source_impl.cc.o -c /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib/pluto_source_impl.cc
[6/24] /usr/bin/x86_64-pc-linux-gnu-g++ -DGR_VERSION_3_7_OR_LESS -Dgnuradio_iio_EXPORTS -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/include -Ilib -Iinclude  -DNDEBUG -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fPIC   -fvisibility=hidden -MD -MT lib/CMakeFiles/gnuradio-iio.dir/fmcomms5_sink_impl.cc.o -MF lib/CMakeFiles/gnuradio-iio.dir/fmcomms5_sink_impl.cc.o.d -o lib/CMakeFiles/gnuradio-iio.dir/fmcomms5_sink_impl.cc.o -c /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib/fmcomms5_sink_impl.cc
[7/24] /usr/bin/x86_64-pc-linux-gnu-g++ -DGR_VERSION_3_7_OR_LESS -Dgnuradio_iio_EXPORTS -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/include -Ilib -Iinclude  -DNDEBUG -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fPIC   -fvisibility=hidden -MD -MT lib/CMakeFiles/gnuradio-iio.dir/pluto_sink_impl.cc.o -MF lib/CMakeFiles/gnuradio-iio.dir/pluto_sink_impl.cc.o.d -o lib/CMakeFiles/gnuradio-iio.dir/pluto_sink_impl.cc.o -c /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib/pluto_sink_impl.cc
[8/24] /usr/bin/x86_64-pc-linux-gnu-g++ -DGR_VERSION_3_7_OR_LESS -Dgnuradio_iio_EXPORTS -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/include -Ilib -Iinclude  -DNDEBUG -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fPIC   -fvisibility=hidden -MD -MT lib/CMakeFiles/gnuradio-iio.dir/device_sink_impl.cc.o -MF lib/CMakeFiles/gnuradio-iio.dir/device_sink_impl.cc.o.d -o lib/CMakeFiles/gnuradio-iio.dir/device_sink_impl.cc.o -c /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib/device_sink_impl.cc
[9/24] /usr/bin/x86_64-pc-linux-gnu-g++ -DGR_VERSION_3_7_OR_LESS -Dgnuradio_iio_EXPORTS -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib -I/var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/include -Ilib -Iinclude  -DNDEBUG -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fPIC   -fvisibility=hidden -MD -MT lib/CMakeFiles/gnuradio-iio.dir/fmcomms2_source_impl.cc.o -MF lib/CMakeFiles/gnuradio-iio.dir/fmcomms2_so [... cut ...] s -fPIC   -fvisibility=hidden -MD -MT lib/CMakeFiles/gnuradio-iio.dir/attr_source_impl.cc.o -MF lib/CMakeFiles/gnuradio-iio.dir/attr_source_impl.cc.o.d -o lib/CMakeFiles/gnuradio-iio.dir/attr_source_impl.cc.o -c /var/tmp/portage/net-wireless/gr-iio-0.3_p20191029/work/gr-iio-8196f3de5b2fdf384801748166cc4ad86a4a9338/lib/attr_source_impl.cc
ninja: build stopped: subcommand failed.
 * ERROR: net-wireless/gr-iio-0.3_p20191029::maintainer-zero failed (compile phase):
 *   ninja -v -j12 -l12 failed
```

tfcollins, 2020-06-01:
See here: https://wiki.analog.com/resources/tools-software/linux-software/gnuradio

-Travis

ZeroChaos-, 2020-06-02:
thanks, tested and working.  I'm going to warmly suggest that master should support gnuradio 3.8 at this point, and this branch should be merged into master.  Thanks for your hard work.

closed 2020-06-02 by ZeroChaos-