analogdevicesinc/gr-iio #91: make:  boost error

black-pigeon, 2020-11-12:
[image]
Hello， When I started to work with this project， the problem shows in the picture，what could result this？ Thanks a lot!!!

black-pigeon, 2020-11-12:
When I get back to the last version, there shows no error. Maybe the latest version have some problem? Or maybe my computer environment need some changes?

commodo, 2020-11-12:
What version of the boost libraries are you using?
And what distro?

kerrynwood, 2020-11-29:
I think this is related to #87.  I had a similar problem on Mac (clang V12.0.0), compiling against boost 1.60.

My fix was to add the boost:placeholders namespace.

`diff --git a/lib/attr_sink_impl.cc b/lib/attr_sink_impl.cc
index 0bc869c..ae088bb 100644
--- a/lib/attr_sink_impl.cc
+++ b/lib/attr_sink_impl.cc
@@ -29,6 +29,8 @@
 #include <iostream>
 #include <boost/lexical_cast.hpp>

+using namespace boost::placeholders;
+
 namespace gr {
   namespace iio {`

closed 2021-09-25 by black-pigeon

u-eng, 2021-11-03:
Adding 'namespace boost::placeholders' in attr_sink_impl.cc doesn't seem to work for me. I'm on Ubuntu 16.04, installed gnuradio 3.7.9.1 using apt install gnuradio, and cloned the master branch of gr-iio (which is same as maint-3.7 I guess). Any ideas on how you resolved this issue?