analogdevicesinc/gr-iio #95: Does not build with GNU Radio 3.9

rgov, 2021-01-21:
GNU Radio 3.9 moved from SWIG to PyBind11, which might explain this build error (on macOS):

```
==> cmake .. -DPYTHON_EXECUTABLE=/usr/local/Cellar/gr-iio/HEAD-d0a4adc/libexec/venv/bin/python
Last 15 lines from /Users/rzg/Library/Logs/Homebrew/gr-iio/02.cmake:
-- Found BISON: /usr/local/opt/bison/bin/bison (found suitable version "3.7.4", minimum required is "3.0.2") 
-- Found SWIG: /usr/local/bin/swig (found version "4.0.2")  
-- Found PythonLibs: /usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/libpython3.9.dylib (found version "3.9.1") 
CMake Error at swig/CMakeLists.txt:36 (include):
  include could not find load file:

    GrSwig

CMake Error at swig/CMakeLists.txt:47 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".
```

rgov, 2021-01-21:
Commit that removed SWIG from GNU Radio: https://github.com/gnuradio/gnuradio/commit/9f6086161f8693b0643215f7a3935ec13661c882

adamhorden, 2021-01-21:
I have this fixed in:

https://github.com/analogdevicesinc/gnuradio/pull/4

💥 

This needs to be merged into gr-iio. I created the PR on the GNU Radio fork as IIO was supposed to be upstreamed in  GNU Radio Release v3.9.0.0 but this has not happened.

The changes from that need to be merge into gr-iio as the OOT module. 

I will create a PR to bring my changes across 🤓 .

rgov, 2021-01-21:
Awesome. Do you know what is going on with `master` vs `upgrade-3.8` here?

adamhorden, 2021-01-22:
@rgov I am not sure on that. I am just an end user, using GNU Radio with Analogue Devices Plutos.

I have started a work in progress pull request here:

https://github.com/analogdevicesinc/gr-iio/pull/96

anilgurses, 2021-06-03:
Is there any update?

adamhorden, 2021-06-03:
@anilgurses this has been superseded by:

https://github.com/gnuradio/gnuradio/pull/4277

We are currently working on bringing `gr-iio` in tree. If you compiled that branch from source you would get `gr-iio` support. We are getting closer every day now to merging this in. Lots of work has happened in the `feature/gr-iio` branch here 🤓 :

https://github.com/gnuradio/gnuradio/tree/feature/gr-iio

**Adam Horden**