analogdevicesinc/gr-iio #97: Make: error gr::iio

mjihad-75, 2021-01-29:
Hi,
I am trying to compile gr-iio in version 3.8 (I have cloned the repository from the upgrade-3.8 branch) and I get the following error when doing make:

[image]

Can anyone help me? Thank you very much!

pgoettle, 2021-02-15:
i have the same issue.  I was looking at what Adam Holden posted in Issue #96 (there is a link to this issue in his Issue # 95 posting) and it seems that he is on track to address the incompatibility issue between GNU radio 3.9 and gr-iio.  I'm not a programming expert - barely a novice - but it seems that gr-iio is using the boost libraries for pointers or shared pointers (I'm sure I'm not articulating that properly) and GNU radio 3.9 is using a later version of C++ that has this capability built-in.  

A C++ expert can articulate this much better.

pgoettle, 2021-02-16:
Sorry - i have Mr. Horden's name wrong.  It's Adam Horden