analogdevicesinc/gr-iio #99: Can't build the latest gr-iio

ahmedshahein, 2021-01-30:
I have upgraded GNURadio to 3.9 and I am trying to install the Pluto modules but it is not going smoothly.
I have Ubuntu 18.04.
When I execute the command:
~/Downloads/cmake-3.8.2-Linux-x86_64/bin/cmake .

I get
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
CMake Error at cmake/Modules/GrSwig.cmake:188 (GR_LIBTOOL):
  Unknown CMake command "GR_LIBTOOL".
Call Stack (most recent call first):
  swig/CMakeLists.txt:54 (GR_SWIG_INSTALL)

-- Configuring incomplete, errors occurred!

I also tried CMAKE 3.19 with no luck.

Could you please advise on that issue?
regards.

ahmedshahein, 2021-01-30:
Each CMake has different error

CMake 3.19:
CMake Warning at CMakeLists.txt:120 (find_package):
  By not providing "FindIIO.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "IIO", but
  CMake did not find one.

  Could not find a package configuration file provided by "IIO" with any of
  the following names:

    IIOConfig.cmake
    iio-config.cmake

  Add the installation prefix of "IIO" to CMAKE_PREFIX_PATH or set "IIO_DIR"
  to a directory containing one of the above files.  If "IIO" provides a
  separate development package or SDK, be sure it has been installed.

CMake Warning (dev) at /home/shahein/Downloads/cmake-3.19.4/Modules/FindPackageHandleStandardArgs.cmake:426 (message):
  The package name passed to `find_package_handle_standard_args` (VOLK) does
  not match the name of the calling package (Volk).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/volk/VolkConfig.cmake:32 (find_package_handle_standard_args)
  CMakeLists.txt:121 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:143 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!

CMake 3.8
-- Build type not specified: defaulting to release.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing:  MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
CMake Error at /home/shahein/Downloads/cmake-3.8.2-Linux-x86_64/share/cmake-3.8/Modules/CMakeFindDependencyMacro.cmake:35 (message):
  Invalid arguments to find_dependency
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake:33 (find_dependency)
  CMakeLists.txt:76 (find_package)

-- Configuring incomplete, errors occurred!

Which CMake version shall I use?

regards

tfcollins, 2021-01-30:
gr-iio has not been upgraded to support 3.9 yet, so it may take some time to address these issues. If you wanted to do it on your own I would suggest looking at the common changes that are made to other OOTs that have been already upgraded.

-Travis

FJTor, 2022-09-19:
Hello @ahmedshahein, did you manage to to install gr-iio on Ubuntu 18.04 and gnuradio 3.9?

thanks

tfcollins, 2022-09-19:
@FJTor 3.9 is not supported. Either use 3.8 or 3.10

bezborodow, 2024-10-28:
I have the same issue with GNU Radio 3.10.9.2.

tfcollins, 2024-10-28:
gr-iio is provided in-tree with GNU Radio 3.10. This repo does not support 3.10

closed 2024-10-28 by tfcollins