analogdevicesinc/gr-iio #101: Auto filter option for FMCOMMS source block missing

felixei, 2021-03-23:
Hi there,
Is there a particular reason for the FMCOMMS5 soure block not having the "auto filter" option to allow for lower sample rates than 2.048MSPS, like the FMCOMMS2/3/4 source block has?
In the YAML/XML block descriptor the assertion for the 'auto filter' option is still there (maybe just a leftover?) but apart form that the 'auto filter' option itself is missing, as well as the final argument for that option in the block's function call.

Does this option exist in some branch already, or Is there a quick workaround for this?
I was trying to geht the sample rate to exactly 2MSPS for ADS-B reception. I could possibly try to design my own filter specifially for that use case but I do not have much experience with it.

I saw that this was also an issue for IIO Oscilloscope reported by Travis in early 2019.

Thanks
Felix

felixei, 2021-03-28:
I am wondering if, as a workaround, I could use two FMCOMMS2 source blocks instead which allow to enable the auto filter option on each single transceiver separately. Is there a way to speicify the IIO device name in the context URI to distinguish between AD9361 chip one and two?

Felix

tfcollins, 2021-03-29:
autofilter uses this function call: http://analogdevicesinc.github.io/libad9361-iio/group__TopLevel.html#ga56c31cc700d410be9bdeea3040e62eb3
You would have to call it for each PHY device (ad9361-phy and ad9361-phy-B). pyadi-iio implements these functions in pure python: https://github.com/analogdevicesinc/pyadi-iio/blob/1005113c1c1712cbcf8daedaf5c3a48e30716213/adi/fmcomms5.py#L190

-Travis

felixei, 2021-03-30:
Hi Travis, thanks for getting back.
I simply took the filter coefficients from the pyadi-iio reference (lines 201-211) and loaded them to the board via .ftr file right through the GNURadio flowgraph - and it worked like a charm. Thanks a lot!

Do you still happen to have the frequency and phase response for these for reference? I think I calculated them properly for 2 MHz sampling frequency, but a comparison would come in handy.

Felix

tfcollins, 2021-03-30:
Here is the normalized response:

[image]