analogdevicesinc/gr-iio #103: CMake Error: The following variables are used in this project, but they are set to NOTFOUND.

skyluke86, 2021-05-24:
Hello,
I'm trying to compile with ubuntu 18.04 but during compile i have got this error:
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
AD9361_INCLUDE_DIRS
   used as include directory in directory /home/devs/Tool/gr-iio/lib
   used as include directory in directory /home/devs/Tool/gr-iio/lib
   used as include directory in directory /home/devs/Tool/gr-iio/lib
   used as include directory in directory /home/devs/Tool/gr-iio/lib
   used as include directory in directory /home/devs/Tool/gr-iio/lib
   used as include directory in directory /home/devs/Tool/gr-iio/lib
   used as include directory in directory /home/devs/Tool/gr-iio/lib
AD9361_LIBRARIES
    linked by target "gnuradio-iio" in directory /home/devs/Tool/gr-iio/lib

-- Configuring incomplete, errors occurred!
See also "/home/devs/Tool/gr-iio/CMakeFiles/CMakeOutput.log".
See also "/home/devs/Tool/gr-iio/CMakeFiles/CMakeError.log".

Can you help me?
Thanks in advance

tfcollins, 2021-05-24:
You are missing dependencies. Please see install steps here: https://wiki.analog.com/resources/tools-software/linux-software/gnuradio

-Travis

tfcollins, 2021-06-05:
Is this still an issue?

-Travis

skyluke86, 2021-06-05:
Thanks for the help it solved the issue :)

closed 2021-06-05 by skyluke86