analogdevicesinc/gr-iio #104: Error when building on Fedora 34

ingframin, 2021-06-10:
Hello,
I am having some errors when building on Fedora 34 / gnuradio 3.9.
I generated the makefiles with Cmake, it was a bit hard, but managed to get all the dependencies in place.

However, when I run make I get the following:

`$ make
[  4%] [BISON][parser] Building parser with bison 3.7.4
[  8%] [FLEX][lexer] Building scanner with flex 2.6.4
Scanning dependencies of target gnuradio-iio
[ 13%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o
/home/user/gr-iio/lib/device_source_impl.cc: In static member function ‘static gr::iio::device_source::sptr gr::iio::device_source::make(const string&, const string&, const std::vector<std::__cxx11::basic_string<char> >&, const string&, const std::vector<std::__cxx11::basic_string<char> >&, unsigned int, unsigned int)’:
/home/user/gr-iio/lib/device_source_impl.cc:45:9: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::iio::device_source_impl]()’ from ‘std::shared_ptr<gr::iio::device_source_impl>’ to ‘gr::iio::device_source::sptr’ {aka ‘boost::shared_ptr<gr::iio::device_source>’}
   44 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
   45 |         (new device_source_impl(device_source_impl::get_context(uri), true,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         std::shared_ptr<gr::iio::device_source_impl>
   46 |                                 device, channels, device_phy,
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   47 |                                 params, buffer_size, decimation));
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/gr-iio/lib/device_source_impl.cc: In static member function ‘static gr::iio::device_source::sptr gr::iio::device_source::make_from(iio_context*, const string&, const std::vector<std::__cxx11::basic_string<char> >&, const string&, const std::vector<std::__cxx11::basic_string<char> >&, unsigned int, unsigned int)’:
/home/user/gr-iio/lib/device_source_impl.cc:58:9: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::iio::device_source_impl]()’ from ‘std::shared_ptr<gr::iio::device_source_impl>’ to ‘gr::iio::device_source::sptr’ {aka ‘boost::shared_ptr<gr::iio::device_source>’}
   57 |       return gnuradio::get_initial_sptr
      |              ~~~~~~~~~~~~~~~~~~~~~~~~~~
   58 |         (new device_source_impl(ctx, false, device, channels, device_phy,
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         std::shared_ptr<gr::iio::device_source_impl>
   59 |                                 params, buffer_size, decimation));
      |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [lib/CMakeFiles/gnuradio-iio.dir/build.make:93: lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:225: lib/CMakeFiles/gnuradio-iio.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
`
The system correctly works with Pluto SDR using SoapySDR in my C++ code.
The problem is when building gr-iio.
Any idea about what it could be?

wheberth, 2021-07-24:
I am also having problems building it on Fedora 34. I have checked out the branch "cmake-update-3.8" as it explicitly states that GR 3.9 should be supported. However cmake complains about not finding GrSwig:

`CMake Error at swig/CMakeLists.txt:36 (include):
  include could not find requested file:

    GrSwig

CMake Error at swig/CMakeLists.txt:47 (GR_SWIG_MAKE):
Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!`

tfcollins, 2021-07-26:
gr-iio 3.8 branches do not support 3.9 since it switched from swig to pybindC++.  Currently 3.9 is not supported.

mobilinkd, 2021-12-31:
Will gr-iio ever support GNU Radio 3.9?

computer-wizard, 2022-01-01:
Version 3.10.0.0 will have it built-in  See https://github.com/gnuradio/gnuradio/releases/tag/v3.10.0.0-rc1
The stable release is planned for the end of January they say.

mobilinkd, 2022-01-03:
I understand.  But there are distributions that will be around for at least another year or two that ship with GNU Radio 3.9.  The ADALM Pluto is currently unusable on those platforms.

computer-wizard, 2022-01-03:
I was in the same situation around spring of last year as I couldn't use my PlutoSDR  anymore after an upgrade. Luckily as of GnuRadio 3.9.4.0 (or maybe even a little bit earlier) there is support for the Soapy blocks, including the Pluto.  You have to compile the  SoapyPlutoSDR hardware driver yourself.  You can download the source from https://github.com/pothosware/SoapyPlutoSDR  You may also read https://github.com/pothosware/SoapyPlutoSDR/issues/45  of how to increase the buffer size in PlutoSDR_Streaming.cpp at line 176 & 610. I changed it to 64k ( 65536) to get the highest throughput (and sample rate) I also installed the Soapy Server/Remote locally. With that I could even get an higher sample rate as the Soapy Server  can convert from CS 8 to CS 16 . In the Soapy Pluto Sink of GnuRadio I use, 'format=CS8,driver=remote,remote=tcp://127.0.0.1:55132,remote:driver=plutosdr' for the device arguments.  tcp port has to be 
adapted to the local situation,  For me this gives even an higher throughput/SR as with gr-iio