analogdevicesinc/gr-iio #111: Cross comiling gr-iio shows error -lpthreads

smx12, 2022-02-01:
I'm trying to cross compile gr-iio from source to ARM. But it seems to be a typo with the parameter lpthreads I think it should be lpthread (without the "s"). I haven't been able to find where this parameter is added. I'll attach CMakeError.log

```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/user/pluto/gr-iio/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_00088/fast && /usr/bin/make -f CMakeFiles/cmTC_00088.dir/build.make CMakeFiles/cmTC_00088.dir/build
make[1]: se entra en el directorio '/home/user/pluto/gr-iio/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_00088.dir/src.c.o
/usr/bin/arm-linux-gnueabihf-gcc   -O3 -mfloat-abi=hard --sysroot=/home/user/pluto/pluto-0.34.sysroot/ -DCMAKE_HAVE_LIBC_PTHREAD   -std=gnu11 -o CMakeFiles/cmTC_00088.dir/src.c.o   -c /home/user/pluto/gr-iio/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_00088
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_00088.dir/link.txt --verbose=1
/usr/bin/arm-linux-gnueabihf-gcc -O3 -mfloat-abi=hard --sysroot=/home/user/pluto/pluto-0.34.sysroot/ -DCMAKE_HAVE_LIBC_PTHREAD    CMakeFiles/cmTC_00088.dir/src.c.o  -o cmTC_00088 
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: CMakeFiles/cmTC_00088.dir/src.c.o: en la función `main':
src.c:(.text.startup+0x28): referencia a `pthread_create' sin definir
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: src.c:(.text.startup+0x2e): referencia a `pthread_detach' sin definir
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: src.c:(.text.startup+0x36): referencia a `pthread_join' sin definir
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_00088.dir/build.make:87: cmTC_00088] Error 1
make[1]: se sale del directorio '/home/user/pluto/gr-iio/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_00088/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/user/pluto/gr-iio/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_93028/fast && /usr/bin/make -f CMakeFiles/cmTC_93028.dir/build.make CMakeFiles/cmTC_93028.dir/build
make[1]: se entra en el directorio '/home/user/pluto/gr-iio/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_93028.dir/CheckFunctionExists.c.o
/usr/bin/arm-linux-gnueabihf-gcc   -O3 -mfloat-abi=hard --sysroot=/home/user/pluto/pluto-0.34.sysroot/ -DCHECK_FUNCTION_EXISTS=pthread_create   -std=gnu11 -o CMakeFiles/cmTC_93028.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_93028
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_93028.dir/link.txt --verbose=1
/usr/bin/arm-linux-gnueabihf-gcc -O3 -mfloat-abi=hard --sysroot=/home/user/pluto/pluto-0.34.sysroot/ -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_93028.dir/CheckFunctionExists.c.o  -o cmTC_93028  -lpthreads 
/usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: no se puede encontrar -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_93028.dir/build.make:87: cmTC_93028] Error 1
make[1]: se sale del directorio '/home/user/pluto/gr-iio/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_93028/fast] Error 2
```
Is this a bug or am I doing something wrong? Thanks!

tfcollins, 2022-02-02:
I don't believe we explicitly link against pthreads in our CMake so it likely comes from GR Runtime or the toolchain updates you are adding. Did you have a similar issue when building GR itself as they should be using the same recipe if not from you?

-Travis

smx12, 2022-02-03:
I fixed it, there were problems with libraries on my PC, I found it out when I tried to do a native compile and it gave me the same errors. Thanks!

closed 2022-02-03 by smx12