analogdevicesinc/gr-iio #112: GR 3.8: Unknown CMake command "GR_PYTHON_CHECK_MODULE"

warnes, 2022-03-30:
After manually installing MPIR and specifying the appropriate paths for cmake, it generates this error:
```
$ cmake ..
-- Build type not specified: defaulting to release.
-- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.35") found components: filesystem system thread date_time chrono 
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Found MPIR: /usr/local/lib/libmpir.so  
-- Found Boost: /usr/lib/aarch64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
CMake Error at /usr/lib/aarch64-linux-gnu/cmake/gnuradio/FindTHRIFT.cmake:70 (GR_PYTHON_CHECK_MODULE):
  Unknown CMake command "GR_PYTHON_CHECK_MODULE".
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /usr/lib/aarch64-linux-gnu/cmake/gnuradio/gnuradio-runtimeConfig.cmake:24 (find_dependency)
  /usr/lib/aarch64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake:54 (include)
  CMakeLists.txt:123 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/inoui/src/gr-iio/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/inoui/src/gr-iio/build/CMakeFiles/CMakeError.log".
```

Solution found at [https://github.com/ptrkrysik/multi-rtl/issues/6:](https://github.com/ptrkrysik/multi-rtl/issues/6#issuecomment-1008708085):  Basically change the required `cmake` version, after which `cmake` succeeds.

## System information
```
$ uname -a
Linux wwi-red 5.15.30-v8+ #1536 SMP PREEMPT Mon Mar 28 13:53:14 BST 2022 aarch64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

tfcollins, 2022-03-30:
I'm a little confused. How does the referenced solution actually do anything on your system? It just changes the minimum CMake required but not the version used.

The error seemed to be related to Thrift and not **mpir** as mpir was found on your system.

warnes, 2022-04-01:
I'm not certain, but it did fix the issue.

I suspect that either cmake does things 'backward compatible' if it detects and old version of cmake is specified, or the necessary module checks the requested version and doesn't load if it's too old.  <shrug>

IcingTomato, 2023-11-09:
Thank you very much @warnes ! It works on my raspberry pi!