analogdevicesinc/gr-iio #113: cmake not working for gr-iio

lifeatthesharpend, 2022-04-02:
on WSL2 after installing and building libad9361-iio and libiio as instructed on https://wiki.analog.com/resources/tools-software/linux-software/gnuradio
 gr-iio give error when commanded  "cmake ."

here is the error: 
CMake Error at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake:13 (find_dependency)
  CMakeLists.txt:123 (find_package)

tfcollins, 2022-04-04:
Can you provide the full output of the cmake command from a clean build folder?

-Travis

lifeatthesharpend, 2022-04-04:
/mnt/c/Projects/adi/gr-iio$ cmake .
-- Build type not specified: defaulting to release.
CMake Warning at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake:13 (find_dependency)
  CMakeLists.txt:90 (find_package)

CMake Warning at CMakeLists.txt:90 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency spdlog could not be found.

CMake Warning at /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake:13 (find_dependency)
  CMakeLists.txt:92 (find_package)

CMake Warning at CMakeLists.txt:92 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency spdlog could not be found.

CMake Error at CMakeLists.txt:94 (message):
  Unable to find GNURadio

-- Configuring incomplete, errors occurred!
See also "/mnt/c/Projects/adi/gr-iio/CMakeFiles/CMakeOutput.log".

tfcollins, 2022-04-04:
What version of gnuradio do you have installed? How did you install it?

-Travis

lifeatthesharpend, 2022-04-04:
sudo apt install gnuradio

tfcollins, 2022-04-05:
You are likely missing the development packages. These are required for all OOT modules:
```
sudo apt install gnuradio-dev
```